ovro-lwa/lwa-issues #436: schedule software deployment

Princemongo, 2023-06-28:
On 6/27 I updated the system configuration spreadsheet. @rh-codebase pushed this into Git, but the rfimitigation plots do not reflect this change. The Python script on gpu03 which creates a single plot, genplot.py, displays the following, which should have read the most recent version, but does not. This is what is displayed:

"Read antpos from xlsx file in repo
/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/lwautils/conf/etcdConfig.yml"

I am not sure why this does not work. @caseyjlaw - do you know how to fix this?

caseyjlaw, 2023-06-28:
The latest data would be in etcd and the logging statement should say that (instead of xlsx, which is a fallback option).
There was a version of the `lwa-antpos` code that could not parse etcd correctly and would fall back to xlsx. I suspect you have old code.
I suggest running `conda activate deployment` to get the latest code (it gets set up in a different software environment).

rh-codebase, 2023-06-28:
there is no 'deployment' conda environment on gpu03.

caseyjlaw, 2023-06-28:
Ah, so there isn't. I see it on gpu09, so @Princemongo could move there.
However, we are encouraging users to move analysis to cal-im nodes to avoid conflict with real-time systems and avoid software issues such as this. I'd suggest going to lwacalim10.mcs.pvt for this sort of analysis. If you don't have an account, then you could make a request to Kemby.

caseyjlaw, 2023-08-02:
@rh-codebase Can we add lxdlwagpu09 to the list of servers that get software deployment?

caseyjlaw, 2023-10-12:
I'll close this, as the original issue is resolved and we have a way to deploy software on gpu servers.

closed 2023-10-12 by caseyjlaw