ovro-lwa/lwa-issues #437: Monitoring data collection and access

greghell, 2023-06-28:
Here is a tentative list of monitoring data:

F-engine
* autocorrelation spectra (if possible?) (float)
* bit stats (mean and RMS of ADC streams) (float)
* PFB overflow count (int)
* PFB clip count (int)

ARX
* current (FEE or photodiode) (float)
* ARX power (float)
* 'sig_on' (bool)
* 'narrow_lpf' (bool)
* 'narrow_hpf' (bool)
* 'first_atten' (float)
* 'second_atten' (float)
* 'dc_on' (bool)

We will need access to these data for the following user cases:

1. real-time data access (directly through ETCD?) for maintenance
2. historical data visualization (grafana?) for data exploration after observing
3. historical data access for misc processing (e.g. estimate flagger threshold)

rh-codebase, 2023-06-28:
at what cadence? Is there a use case for occasional high cadence data? The standard monitor spec is 1s cadence.

caseyjlaw, 2023-07-27:
We will create two new modules in `mnc-python`: "influx.py" and "mpflagging.py". These will expose monitor point time series and use them to create boolean masks, respectively.

greghell, 2023-07-27:
Current monitoring points:

Columns in Influx for the following subsystems:

ARX data in Influx:
time                ant_num arx_addr_a arx_addr_a_1 arx_addr_b arx_addr_b_1 arx_chan_a arx_chan_a_1 arx_chan_b arx_chan_b_1 atten1_a atten1_b atten2_a atten2_b brd_current_a brd_current_b brd_temp_a brd_temp_b current_a current_b dc_on_a dc_on_b narrow_hpf_a narrow_hpf_b narrow_lpf_a narrow_lpf_b pol_a pol_b power_a            power_b            sig_on_a sig_on_b temp_a temp_b

WX data:
time                 abshumid airtemp dwpoint pressure rainhr rainrate raintoday rainyr relhumid saecorrectionfactor vaporpressure watercolumn watervapordensity windchill winddir windspeed

Shelter A/C data:
time                airtemp dwpoint relhumid

Shelter Environmental data:
time                airtemp dwpoint relhumid

Shutdown Status data:
time                currenttemp hostname mediannumsamples mediantemp shutdowntemp status                             statusstate

caseyjlaw, 2023-07-27:
@greghell I have an example function to get influx data in the cjl/dev branch of `mnc-python`.

```
> from mnc import influx
> df = influx.get_rfpower('2023-07-23T00:00:00', 100)
> df
                                 ant_num         pa          pb
2023-07-23 00:00:51.727340+00:00       1  47.590720  102.040816
2023-07-23 00:00:51.727340+00:00      10  62.159308   92.328425
2023-07-23 00:00:51.727340+00:00     100  83.101652   83.101652
2023-07-23 00:00:51.727340+00:00     101  78.670374   70.172031

```

The output has a row for each time/ant. An alternative would be to have a row that is a time sample, which has a 2d matrix with axes of time and ant.
What would you prefer for your application?

greghell, 2023-07-27:
That's great! Let's keep it like that for now. Thank!