ovro-lwa/lwa-issues #438: antenna 55 possible polarization swap

rh-codebase, 2023-06-28:
Grafana plots of ARX RF power shows a mismatch in power levels for spikes received by most antennas in their polarizations. Antenna 55 pol B spike power is consistent with most other antennas pol A spike power. The channel current however is not inconsistent. @Princemongo notes that coax pol A boards have an LED and pol B boards do not. Hence the channel current for pol A signals will be 20-30mA higher. This current difference is consistent with Ant 55 so still a mystery.

Princemongo, 2023-06-28:
The currents for polarizations indicated that the polarizations were correct. Charlie went out to the antenna and found that the antenna had rotated ~ 70-80 degrees! He is going to re-align the antenna and clamp it so that it won't rotate.
[LWA-55 June 2023.pdf](https://github.com/ovro-lwa/lwa-issues/files/11900314/LWA-55.June.2023.pdf)

Princemongo, 2023-06-30:
Charlie aligned and clamped the antenna.

closed 2023-09-08 by yupinghuang