ovro-lwa/lwa-issues #441: LWA Triggering issues

nkosogor, 2023-07-14:
Hi @jack-h,
@caseyjlaw and I have been testing lwa triggers and have come across some strange behavior when a trigger event occurs. 

The primary points of concern are:

1. It appears that the pipeline is writing out one buffer and then stopping (nfile was =1 and ntime_per_file=~40e6). The resulting files turned out to be smaller than what we need. Instead of 30 minutes, it wrote out about 3 minutes.

2. Is it the right approach to create multiple files with nfile>1. If that’s the case, then what ntime_per_file should be set to? Is it just a default value 1e6 from https://github.com/realtimeradio/caltech-bifrost-dsp/blob/d1df3b0f2fa6dfd8d0356d30c7731e4e271cdad8/pipeline/lwa352_pipeline/blocks/triggered_dump_block.py#L155 (we noticed that the first file had ntime of ~4e6, which is inconsistent with the default `ntime_per_file` value of 1e6, but at the same time is not equal to the one that was sent) 

3. Casey attempted to trigger several times but only the first one records data. When subsequent triggers were sent, no additional data was recorded. The x-engines seemed to be found in a “waiting” state.

caseyjlaw, 2023-07-14:
The "waiting" state was noticed once, but could not be reproduced. On the second test, I could trigger once (again writing only one file, despite nfiles>1) and subsequent trigger commands to the pipeline were ignored (or perhaps produced an unseen error).

caseyjlaw, 2023-07-14:
I'm looking at the logs for one of the pipelines. I can see that one trigger signal is seen, but no logging that it completed or indication of the second command sent:
```
2023-07-14 21:50:52 [INFO    ] TRIGGERED DUMP >> Opening /data1/lwa-dump-1689371451.96.tbf.0
```

jack-h, 2023-07-20:
Can you share the sequence of commands that got here? It may also be important which nodes are being triggered, in case the issue is some local hardware configuration issue.

I don't know how big a file is reasonable, but probably trying to store 30 mins of data in a single ~TB file isn't the best approach. I'd suggest an `n_time_per_file` which gives rise to files a few GB in size. Each time sample is 352 stations x 2 polarization x 96 freq chans = 67 kBytes

jack-h, 2023-07-27:
`ntime_per_file` of 1000000 should yield files of 67GB, which seems big, but should work.

Looking at the GPU nodes, I don't see any of them mounting the NVME buffers at `/data0` or `/data1`, so I'd imagine you are triggering to the (single) OS drive which can't sustain the data rate. 

I've manually mounted the partitions (which aren't even named consistently over GPU nodes):
```
# works for some nodes
sudo mount /dev/md126 /data0
sudo mount /dev/md127 /data1
# works for other nodes
sudo mount /dev/md0 /data0
sudo mount /dev/md1 /data1
```

Does this make everything magically work?

@rh-codebase Is the right way to set up the mounts to simply modify the provisioning's `/etc/fstab` population? Any ideas about why the RAID volumes are getting different names on different nodes?

rh-codebase, 2023-07-27:
gpu02, 03 look to be the outliers. those were the ones used to debug the prov. scripts. They were not redone  using the scripts. 04-08 used the scripts and 01 is consistent with those. no idea why the names are different. The script uses /dev/md0 and /dev/md1 when creating the RAID and XFS filesystem. The script also mounts the devices so have no idea who or what is unmounting them.  01,03 have uptime of 37 days and the rest have a 42 day uptime. Ah. I think the shelter was powered down for a transformer install so that would explain the 42 day uptime, but not the 37 day one. Yes. adding the mounts to /etc/fstab sounds like the way forward unless of course the OS decides to rename the devices on boot. I'll file an issue.

jack-h, 2023-07-28:
Thanks @rh-codebase. I'm going to close this since I think the mounting issue is now separate. Feel free to reopen if anyone thinks this isn't right.

closed 2023-07-28 by jack-h