ovro-lwa/lwa-issues #442: x-engine logs grow too large

caseyjlaw, 2023-07-17:
The lxdlwagpu0x servers are running out of space in their home directory. I see four x-engine pipelines have filled four logs of several GB each that date back 8 months.
We should have a way to cap or roll those logs over.

caseyjlaw, 2023-07-17:
I'll remove them and restart the x-engines for now (for my own purposes).

jaycedowell, 2023-07-28:
The logs don't look like they have rolled on the couple of gpu servers I checked.  I would expected them to by now.  Could this be an environment problem?  The pipelines look to be running under casa.

jaycedowell, 2023-07-28:
Nope, casa seems to have the correct version of `lwa352-pipeline.py`.  Strange.

jaycedowell, 2023-08-01:
They seem to be rolling now.

closed 2023-08-01 by jaycedowell