ovro-lwa/lwa-issues #444: recent fengine shenanigans investigation log

yupinghuang, 2023-07-21:
calim01 log showing what might be the first sign of trouble....do the timestamps (UTC) ring a bell to anyone? Why is there a 12-day gap between Jul 07 and Jul 19?

```
journalctl --unit lwa-feng-etcd@snap01
```

```
Jul 07 10:53:52 lwacalim10 lwa_start_feng_[748406]: 2023-07-07 10:53:52,342 - lwa_f.blocks.block:snap01 - WARNING - eq - Some coefficients out of range
Jul 07 10:53:52 lwacalim10 lwa_start_feng_[748406]: 2023-07-07 10:53:52,375 - lwa_f.blocks.block:snap01 - WARNING - eq - Some coefficients out of range
Jul 07 10:53:52 lwacalim10 lwa_start_feng_[748406]: 2023-07-07 10:53:52,408 - lwa_f.blocks.block:snap01 - WARNING - eq - Some coefficients out of range
Jul 07 10:53:52 lwacalim10 lwa_start_feng_[748406]: 2023-07-07 10:53:52,443 - lwa_f.blocks.block:snap01 - WARNING - eq - Some coefficients out of range
Jul 07 10:53:52 lwacalim10 lwa_start_feng_[748406]: 2023-07-07 10:53:52,477 - lwa_f.blocks.block:snap01 - WARNING - eq - Some coefficients out of range
Jul 07 10:53:52 lwacalim10 lwa_start_feng_[748406]: 2023-07-07 10:53:52,521 - lwa_f.blocks.block:snap01 - WARNING - eq - Some coefficients out of range
Jul 07 10:53:52 lwacalim10 lwa_start_feng_[748406]: 2023-07-07 10:53:52,555 - lwa_f.blocks.block:snap01 - WARNING - eq - Some coefficients out of range
Jul 07 10:53:52 lwacalim10 lwa_start_feng_[748406]: 2023-07-07 10:53:52,588 - lwa_f.blocks.block:snap01 - WARNING - eq - Some coefficients out of range
Jul 19 12:01:42 lwacalim10 lwa_start_feng_[748406]: 2023-07-19 12:01:41,946 - /opt/devel/pipeline/envs/deployment/bin/lwa_start_feng_etcd_service.py:snap01 - ERROR - Command failed
                                                    Traceback (most recent call last):
                                                      File "/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/lwa_f/snap2_feng_etcd_client.py", line 564, in _etcd_callback
                                                        resp = cmd_method(**cmd_kwargs)
                                                      File "/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/lwa_f/blocks/autocorr.py", line 208, in get_new_spectra
                                                        spec = self._read_bram() / float(self.get_acc_len())
                                                      File "/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/lwa_f/blocks/autocorr.py", line 145, in _read_bram
                                                        raw = self.read(ramname, 4*n_words_per_stream)
                                                      File "/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/lwa_f/blocks/block.py", line 214, in read
                                                        return self.host.read(self.prefix + reg, nbytes, **kwargs)
                                                      File "/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/casperfpga/casperfpga.py", line 258, in read
                                                        data = self.transport.read(device_name, size, offset, **kwargs)
Jul 07 10:53:52 lwacalim10 lwa_start_feng_[748406]: 2023-07-07 10:53:52,375 - lwa_f.blocks.block:snap01 - WARNING - eq - Some coefficients out of range
Jul 07 10:53:52 lwacalim10 lwa_start_feng_[748406]: 2023-07-07 10:53:52,408 - lwa_f.blocks.block:snap01 - WARNING - eq - Some coefficients out of range
Jul 07 10:53:52 lwacalim10 lwa_start_feng_[748406]: 2023-07-07 10:53:52,443 - lwa_f.blocks.block:snap01 - WARNING - eq - Some coefficients out of range
Jul 07 10:53:52 lwacalim10 lwa_start_feng_[748406]: 2023-07-07 10:53:52,477 - lwa_f.blocks.block:snap01 - WARNING - eq - Some coefficients out of range
Jul 07 10:53:52 lwacalim10 lwa_start_feng_[748406]: 2023-07-07 10:53:52,521 - lwa_f.blocks.block:snap01 - WARNING - eq - Some coefficients out of range
Jul 07 10:53:52 lwacalim10 lwa_start_feng_[748406]: 2023-07-07 10:53:52,555 - lwa_f.blocks.block:snap01 - WARNING - eq - Some coefficients out of range
Jul 07 10:53:52 lwacalim10 lwa_start_feng_[748406]: 2023-07-07 10:53:52,588 - lwa_f.blocks.block:snap01 - WARNING - eq - Some coefficients out of range
Jul 19 12:01:42 lwacalim10 lwa_start_feng_[748406]: 2023-07-19 12:01:41,946 - /opt/devel/pipeline/envs/deployment/bin/lwa_start_feng_etcd_service.py:snap01 - ERROR - Command failed
                                                    Traceback (most recent call last):
                                                      File "/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/lwa_f/snap2_feng_etcd_client.py", line 564, in _etcd_callback
                                                        resp = cmd_method(**cmd_kwargs)
                                                      File "/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/lwa_f/blocks/autocorr.py", line 208, in get_new_spectra
                                                        spec = self._read_bram() / float(self.get_acc_len())
                                                      File "/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/lwa_f/blocks/autocorr.py", line 145, in _read_bram
                                                        raw = self.read(ramname, 4*n_words_per_stream)
                                                      File "/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/lwa_f/blocks/block.py", line 214, in read
                                                        return self.host.read(self.prefix + reg, nbytes, **kwargs)
                                                      File "/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/casperfpga/casperfpga.py", line 258, in read
                                                        data = self.transport.read(device_name, size, offset, **kwargs)
                                                      File "/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/casperfpga/transport_tapcp.py", line 558, in read
                                                        raise RuntimeError("Failed to read size %d from register %s at offset %d" % (size, device_name, offset))
                                                    RuntimeError: Failed to read size 8192 from register autocorr_common_dout4_bram at offset 0
Jul 19 12:01:42 lwacalim10 lwa_start_feng_[748406]: 2023-07-19 12:01:42,114 - /opt/devel/pipeline/envs/deployment/bin/lwa_start_feng_etcd_service.py:snap01 - ERROR - Responded to command 'get_new_spectra' (ID 1>
Jul 19 20:40:48 lwacalim10 lwa_start_feng_[748406]: 2023-07-19 20:40:48,689 - lwa_f.blocks.block:snap01 - ERROR - powermon - Error configuring INA219 sensor at address 0x40
                                                    Traceback (most recent call last):
                                                      File "/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/lwa_f/blocks/powermon.py", line 139, in initialize
                                                        s.init()
                                                      File "/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/casperfpga/i2c_volt.py", line 349, in init
                                                        self.write(rid, val)
                                                      File "/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/casperfpga/i2c_volt.py", line 369, in write
                                                        self.itf.write(self.addr,reg,[data>>8,data&0xff])
                                                      File "/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/casperfpga/i2c.py", line 378, in write
                                                        raise error
                                                      File "/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/casperfpga/i2c.py", line 370, in write
                                                        self._write(addr,cmd+data)
                                                      File "/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/casperfpga/i2c.py", line 225, in _write
                                                        self._itf_write(commandReg, CMD_WRITE)
                                                      File "/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/casperfpga/i2c.py", line 193, in _itf_write
[... cut ...]
                                                       File "/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/lwa_f/snap2_fengine.py", line 474, in cold_start_from_config
                                                         raise RuntimeError
                                                     RuntimeError

```

From Kathryn

```
I don't know which metric triggered it but I did notice the clock rates listed on the dashboard were all over the place, even though the snap2 boards all reported the right clock rate through the Snap2Fengine direct python interface.
```

yupinghuang, 2023-09-23:
Closing in favor of https://github.com/ovro-lwa/lwa-issues/issues/476

although the error message here looks slightly different.

re clock rate though, from Jack

"The calculation reported on the dashboard is an estimate, so I suppose it's not impossible that slow comms (possibly unreliable comms resulting in TFTP retries) could skew things https://github.com/realtimeradio/caltech-lwa/blob/e928ce0d51b2094385302a0a300262663a900540/control_sw/src/blocks/fpga.py#L39"

caseyjlaw, 2023-12-01:
I am seeing f-engine issues after a power cycle and reprogram/initialization. Things seemed ok, but the clock rate of snaps would not settle down to a single value. They also set their status monitor point `'ok': False`.
I do see this in f-engine logs when trying to do a cold_start_from_config:
`2023-12-01 00:02:23,370 - lwa_f.blocks.block:snap02 - INFO - sync - Skipping telescope time update, because this board doesn't have a PPS`

caseyjlaw, 2023-12-01:
Yuping may have resolved this by lowering priority for other processes on lwaclaim10:
`sudo renice -n -1 -u pipeline`

This would give the f-eng etcd clients higher priority and perhaps lower latency on etcd updates.

jack-h, 2023-12-01:
FWIW, there are other tests one could replace the clock estimate with which would be less sensitive (or not sensitive at all) to etcd performance.
One option is to get the ADC clock rate only from SNAP01, which has a PPS so can count clock ticks between PPS edges. Other boards don't get a PPS but do get a sync pulse, so they can measure clock ticks between sync edges (there should be 536870912). The downside of this test is that if the sample clock is wrong, this test will still pass, since the sync clock is itself made from dividing down the sample clock on snap01.

Basically, the sync system which uses snap01 as the timekeeper for other boards is annoying, and it would, IMO, be much more sensible to just give all boards a PPS.

yupinghuang, 2023-12-07:
sync system lesson learned!

I'll get pipeline things more guaranteed resources after the reboot on 12/16. Meanwhile the clock rates seem quite ok.

The original error message was due to the microblaze being gone.

closed 2023-12-07 by yupinghuang