ovro-lwa/lwa-issues #447: VLBI block failure and triggering problem

nkosogor, 2023-07-26:
@jack-h,
I've been trying to test triggering with these commands today separetly from the relay server:
`from mnc import control`
`p = con.pipelines[1]`
`p.triggered_dump.trigger(ntime_per_file=24000,nfile=1,dump_path='/data0/')`
And I got this error:
```
/opt/devel/pipeline/envs/development/lib/python3.6/site-packages/lwa352_pipeline_control/etcd_control.py in send_command(self, host, pipeline, block, inst_id, cmd, timeout, **kwargs)
    311                 self.ec.cancel_watch(watch_id)
    312                 self.log.error("host %s (pipeline %s) failed to respond to etcd command!" % (host, str(pipeline)))
--> 313                 raise RuntimeError
    314             time.sleep(0.001)
```
Am I making some mistake here?

Then Casey was able to trigger via the relay server but encountered an issue with the VLBI output block after a voltage trigger is sent:
`2023-07-26 22:10:19 [ERROR   ] VLBI OUTPUT >> Sending error: 'NoneType' object has no attribute 'send'`
It seems like the UDPTransmit isn't being initialized as needed ([here](https://github.com/realtimeradio/caltech-bifrost-dsp/blob/d1df3b0f2fa6dfd8d0356d30c7731e4e271cdad8/pipeline/lwa352_pipeline/blocks/beamform_vlbi_output_block.py#L248)) and as a result, it's unable to execute the send command.
I'm unsure why this is happening.

caseyjlaw, 2023-07-26:
The etcd timeout happened when the x-engine process on gpu07 was not behaving well. The home and /data0-1 directories are both on the nvme, which was full. The confusing part is that *I* was able to trigger. 😖 

The VLBI error is a definitely a blocker for us right now.

jaycedowell, 2023-07-27:
What node/pipeline was that `VLBI OUTPUT >> Sending error` from?

caseyjlaw, 2023-07-27:
I think most/all of them showed this error. In a spot check, I see it on 07 at `2023-07-20 00:01:50`.

jaycedowell, 2023-07-27:
Yeah, ok, thanks.  That PR should fix this particular VLBI sending error.

jack-h, 2023-07-28:
Is this closable?

caseyjlaw, 2023-07-28:
There are two issues and the vlbi block is resolved.
I can reproduce the etcd error, but don't understand why it happens. The same syntax works when run by our automated trigger client for ligo events. That client runs under my user account on calim10.

caseyjlaw, 2023-08-02:
The second issue is caused by user error. I'm trying to trigger pipelines with id 0 or 1, but only id 2 or 3 are configured for triggering.

closed 2023-08-02 by caseyjlaw