ovro-lwa/lwa-issues #449: Imaging pipeline Jul 27 + 2 weeks

yupinghuang, 2023-07-27:
- [ ] @yupinghuang to make scripts that generate bandpass cal and give solar folks path
- [ ] @surajit4444mondal to get the solar casa pipeline (which takes one ms and spits out an image) to run under https://github.com/ovro-lwa/distributed-pipeline
- [ ] solar script with infinite main loop goes in to orca/extra/solar @binchensun 
- [ ] @rlbyrne to poke @yupinghuang whenever the new cal scheme is ready for testing somewhat continuously
- [ ] and if we are feeling ambitious... @yupinghuang to make dirty IQUV images semi real-time

yupinghuang, 2023-07-29:
@binchensun @surajit4444mondal See updated README at https://github.com/ovro-lwa/distributed-pipeline/blob/main/README.md

slack msg me if anything looks wrong!

caseyjlaw, 2023-10-26:
Solar imaging script can run on one cal-im node with 10 min cadence and 10 min latency.

yupinghuang, 2023-12-01:
I'm too ashamed to keep this issue going

closed 2023-12-01 by yupinghuang