ovro-lwa/lwa-issues #450: triggering: few files generated on gpu02

nkosogor, 2023-07-27:
@jack-h,
Casey and I have tested triggering with `ntime_per_file=1e6`, and number of files corresponding to about 30 min. It worked great this time and produced 39 files as expected (each of 67 GB) in both `/data0 `and `/data1` directories on all gpu servers except for the second one. For gpu02 `/data0` for some reason seems to have stopped after about 3 min and generated only 5 files.

jack-h, 2023-07-28:
Great!
I can see that gpu02 has only mounted one of its volumes.
I get:
```
(casa) ubuntu@lxdlwagpu02:~$ sudo mount /dev/md0 /data0
mount: /data0: can't read superblock on /dev/md0.
```
I've recreated the RAID array and mounted on `/data0`. I think this should be good to go now. I'm not sure whether this is a real fault or this machine was just never completely set up. Would be useful to re-exercise the provisioning at some point

rh-codebase, 2023-07-28:
thanks for catching 02. scripts can be retested on gpu09 when kathryn is finished with it.  still need to file an issue to script the adding of the mount to /etc/fstab using the uuid.

jack-h, 2023-12-01:
@rh-codebase is it believed that the scripting of these mounts is sorted? If not I'll file/transfer this issue