ovro-lwa/lwa-issues #453: recommendation: use self-correlation spectra for antenna health estimation

ldaddario, 2023-08-02:
Now that visibilities are being recorded continuously, it would be better to use the self-correlation spectra from the X engines to judge antenna health, rather than the spectra computed in the F engines from pre-equalization and pre-requantization data.  The  latter are collected and plotted every 6h by a script running under the crontab owned by 'pipeline'.  Switching to use of the self-correlation spectra has several advantages:
- The cron job is no longer be needed.
- The data are taken from further downstream in the processing, so additional kinds of problems can be detected.
- Spectra from all signals are obtained simultaneously in the same 10s integration.   (The F engine spectra take some time to collect, so those from different SNAP2 boards are not simultaneous.  When there is intermittent RFI, this precludes determining which signals were affected.)
- Spectra are available for any desired time, rather than at 6h intervals.  It is possible to examine variability or look for intermittent effects at 10s cadence.
- Spectra use the same data as are used for imaging, beamforming, and auto-flagging -- rather than something different.
- Since the nominal spectral power has a fixed value that is independent of frequency and signal, deviations from nominal are easier to detect.

To make the data easier for a human to examine, it would be valuable to have a script that extracts all self-correlations from a single integration and plots the spectra.  Since the power can cover a much smaller range than the pre-equalization power, the plots should be linear in power (not logarithmic).  This also makes it easier to see deviations from normal.

caseyjlaw, 2023-08-03:
Related issues #233 #390 #391. But I take Larry's point that there is value in using the *inputs* to flagging algorithms and exposing that to users for quality checks. I think that's a good tool to develop in the context of #391 .

ldaddario, 2023-08-03:
On 8/3/2023 08:13, Casey Law wrote:
> Related issues #233 <https://github.com/ovro-lwa/lwa-issues/issues/233>
> #390 <https://github.com/ovro-lwa/lwa-issues/issues/390> #391
> <https://github.com/ovro-lwa/lwa-issues/issues/391>. But I [...]

I don't think #391 is related, and I'm not sure about the others.  In 
any event, what you "take [as] Larry's point" was not my point at all.

My proposal was only about signal path health estimation.  I said that 
it would be better to use the self-corr spectra from the X engines, 
rather than the pre-equalization spectra from the F engines.  I gave 
multiple reasons for this, none of which had to do with flagging.  While 
*some* flagging algorithms might also use self-corr spectra, that's 
orthogonal to what I was saying.

Many signals with distorted spectra or incorrect power levels can still 
be calibrated, so they remain useful for observing and should (usually) 
not be flagged.  But those spectra still indicate unhealthy signal 
paths.  Conversely, flagging may occur because of RFI, when there is 
nothing unhealthy about the signal path.

Please don't couple this recommendation to issues about flagging.

--Larry

-- 
=======================================================
Larry R. D'Addario
Department of Astronomy
California Institute of Technology
Pasadena, California 91125

291 Cahill, mail code 249-17
email:  ***@***.***
phone:  +1/626/395-1431 (office)
         +1/626/510-6492 (home)
         +1/626/657-6357 (mobile)
=======================================================