ovro-lwa/lwa-issues #456: Simultaneous observing with all beams

iveydavis, 2023-08-03:
When recording started, there seemed to be 3.1% data loss reported across all 8 beams. I think this is consistent with one x-engine pipeline being down though (it remains to be seen that it is the same pipeline across the beams). 

I got quite an extensive collection of warnings that I'm not used to receiving when calibrating the beams. I've attached the output of my calibration command (and my specific calibration command). I haven't looked at them very closely myself yet. They're attached here.
[calibration_output.txt](https://github.com/ovro-lwa/lwa-issues/files/12247474/calibration_output.txt)

jack-h, 2023-08-03:
Errors seem to be `etcd` related, e.g.:

iveydavis, 2023-08-03:
I'll do some more rigorous analysis later, but as a first pass, everything seemed to have gone well! All beams look pretty similar. I think it's safe to use beams 5-8

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]