ovro-lwa/lwa-issues #457: Structure of the fast visibility MS

surajit4444mondal, 2023-08-03:
I think that there is some issue with the structure of the fast visibility MS. WSClean thinks that the MS has 10 spws, instead of 10 scans. WSClean gets stuck if I am try to image the data. CASA can image the data just fine. Marin suggested that we might try to see the effect if we add the keyword "-reorder" to the WSClean call. I tried that, and I found that the image looks like random noise. I am attaching the CASA image and the WSClean image made using "-reorder" with this issue. The data and the code used to generate these images can be found in  /data07/msurajit/fast_vis_support/fast_ms_debug .

/home/surajit/full_sky_casa_fast.png

/home/surajit/full_sky_wsclean_fast.png

dgary50, 2023-08-03:
> /data07/msurajit/fast_vis_support/fast_ms_debug . @caseyjlaw
> @binchensun
> @dgary50

What happens if you ask wsclean to process only one spw (which would be one
time in the actual dataset)?

surajit4444mondal, 2023-08-03:
@dgary50 I am unable to force WSClean to use a single spw. WSClean is supposed to use the spws listed after keyword -spws , but whatever I choose it is doing the exact same thing.

jaycedowell, 2023-08-22:
I looked at this this afternoon and I *think* I know what is going on.  It looks like wsclean doesn't like the 10 rows in DATA_DESCRIPTION.  If I keep only one of row in that table then wsclean seems happy.  I can image either all the data with something like `-field all` or a sub-set with `-interval 4 5 -field all`.  If this change to the MS works for you then I will push it out to the station.

gregghallinan, 2023-08-22:
Nice sleuthing Jayce! Any idea why the show vis imaging won't work in CASA? All other CASA tasks work fine,  but tclean in CASA just produces an image full of zeros, if I recall correctly.

jaycedowell, 2023-08-22:
I was able to reproduce something like that while testing with wsclean when I had data selection criteria that lead to no data matching all of the requirements.  Apart from that I don't have any other ideas.  I can give tclean a shot and see what happens.

dgary50, 2023-08-22:
Great, thanks!  Is this something we can modify for existing (already
recorded) data by editing a table?  If so, please let us know specifically
what change is needed.  We have some fast vis data saved for one of the
type II bursts.

Thanks,
Dale

jaycedowell, 2023-08-23:
Yes, you should be able to modify existing fast measurement sets with something like:
```
from casacore.tables import table
t = table('/your/ms/here/DATA_DECRIPTION', readonly=False)
n = t.nrows()
t.removerows(list(range(1,n)))
t.done()
```
to get down to only a single row in the DATA_DESCRIPTION table.

surajit4444mondal, 2023-08-23:
Thanks Jayce for finding the issue. I am trying it out now and I can make some images. I am testing this now and will update this issue accordingly.

surajit4444mondal, 2023-08-23:
I have verified that I can image all fields, subset of fields, single fields with WSClean if I use the fix suggested by Jayce. So I believe that we can fix this in the DATA writing step itself.

caseyjlaw, 2023-08-23:
> I looked at this this afternoon and I _think_ I know what is going on. It looks like wsclean doesn't like the 10 rows in DATA_DESCRIPTION. If I keep only one of row in that table then wsclean seems [...]

Thanks, Jayce. By 'push it to the station', do you mean you'd like to update the data recorders to write with one row? Or that the fix will be pushed to some imaging script or instructions?

jaycedowell, 2023-08-24:
I tried imaging slow vis. with tclean in CASA 5.8.0 today and I was able to get images out when I used non-zero data (the data I used were from this morning).  I didn't try anything complicated, I only set the image and cell sizes in the call.

closed 2023-09-27 by yupinghuang