ovro-lwa/lwa-issues #460: Interaction between slow vis. and voltage beam outputs

jaycedowell, 2023-08-08:
There seems to be an interaction between the slow visibility and voltage beam outputs from the T-engine.  If I log the packet loss at T-engine I see something like:
[image]
An FFT of that shows a peak near 0.1 Hz (period of 10 s):
[image]

It's not clear what is happening.  At least some of the lost voltage beam packets do eventually make it to the T-engine but after the buffer they are destined for is closed (these are marked as "late" in `like_bmon.py`).

jack-h, 2023-08-09:
What machines are you using the T-engine and slow-vis tests, so I can recreate the traffic flow while poking around switch stats?

jack-h, 2023-08-09:
Notes from today -- I just wrote a dumb C program to receive packets, and timed how long it should take to receive a second's worth of packets. I see some drops (sometimes it takes 1.1 seconds rather than 1s to get the packets) but nothing which looks like it obviously correlates with the 10s cadence of the slow visibilities.
Next I'll try actually interrogating the packet headers.
Nothing on either 40G switch suggests anything is amiss, and I don't see any packets being broadcast. When I turn off volt-beam output, the T-engine shows no packets received via tcpdump.
I did notice that the T-engine CPUs were set to powersave mode. I changed this to performance (before doing so I couldn't get anything like an acceptable packet RX rate)

jaycedowell, 2023-08-09:
I tired to look into your first question with my logs but I'm not sure what to make of the results.  Below I've plotted "late" - "missing" bytes as a function of time:
[image]
The sudden drops are at ~10 s intervals and I interpret those as Bifrost marking the packets "missing" until they show up in the next logging sample as "late".  There is an overall trend of "late" - "missing" decreasing with time which I think means that some packets are never making it.

jack-h, 2023-08-09:
With no slow corr output does the slow downward trend go away? I.e., is the
problem entirely the volt beam + slow corr, or is there another underlying
issue with gradual packet loss.
It certainly looks like the majority of packets are "just" late, and even
the first big drop from 510MB to 508MB doesn't seem to be on a 10s boundary

jaycedowell, 2023-08-09:
I'm not sure on that one.  The only configurations I've tested are slow + voltage and voltage only with no toggling on/off slow.

It entirely possible that the overwhelming majority are just late.  The buffers are 1960 F-engine spectra wide (~81 ms) and two are open at a time.  Maybe the burst of slow output introduces enough scatter in the voltage packet arrival times that the T-engine receiver gets upset?

jack-h, 2023-08-09:
> I'm not sure on that one. The only configurations I've tested are slow +
> voltage and voltage only with no toggling on/off slow.
>
> It entirely possible that the overwhelming majority are just late. [...]

Possible, I guess. Is it easy to increase the buffer sizes? I'm not really
sure how to try to fix things at the TX side if that's not possible.

jaycedowell, 2023-08-09:
I've tried that in the past but that lead to other problems with the pipeline.  It's also not infinitely adjustable because of the re-channelization step.

On the transmit side we could spool the slow data out over a longer period of time to see if that changes things.

jack-h, 2023-08-09:
> I've tried that in the past but that lead to other problems with the
> pipeline. It's also not infinitely adjustable because of the
> re-channelization step.
>
> On the transmit side we could spool the slow data out over a longer period
> of time to see if that changes things.

I'm already doing a bit of that (there's a runtime max mbps parameter) but
I'll play around and see if I can make the issue go away.
The pipelines are also transmitting slow data with random inter-pipeline
delays. Possibly these are just making things worse, so I'll poke around
there too.

jaycedowell, 2023-08-09:
Oh, thanks.  There is definitely some width to the missing fraction spikes in that first plot.  I zoomed in on these and they are typically 3 to 4 seconds wide.  Does that match either the rate limiting or random delay time scale?

jack-h, 2023-08-09:
> Ok, thanks. There is definitely some width to the missing fraction spikes

Think it roughly matches the rate limiting. The random delays are between 0
and 2 seconds

jaycedowell, 2023-08-09:
I played around with the T-engine receiver buffer size.  Below is a comparison between a buffer size of 1960 spectra (left) and 1960*3 = 5880 spectra (right):
[image]
The larger buffer definitely helps but that factor of three growth is as far as I can take things on my end (if I ask for a factor of four it just sits there and does nothing).

jack-h, 2023-08-31:
I've turned off the staggering of slow vis transmitters. I'd be interested to know if this makes things better/worse or has no effect.

jaycedowell, 2023-09-28:
Maybe?  It doesn't seem to be as spiky now but I don't know if that is related to this change or if it's related to work I've been doing on the the T-engine.

jaycedowell, 2023-09-29:
I'm not sure apart from trying to ratchet down the overall output rate and see what that does.  I should also probably try another test without the correlator output running.

jaycedowell, 2023-10-27:
For the record:  things seem different now.  The overall packet drop rate is at ~3-4% and the losses now appear to be sporadic in time.  It might be time to go back to the switch and ask it what is happening.

jaycedowell, 2023-12-07:
I was looking at this again with my T-engine testing and now the pattern is bizarre.  It's like the packet loss is _less_ when slow vis. are being dumped.
[image]