ovro-lwa/lwa-issues #461: data transfer messes with data recorder

yupinghuang, 2023-08-15:
scp/rsync out of the /datax directories seemed to lead the data recorder to report packet drop close to 100%.

gregghallinan, 2023-08-19:
This issue may have happened again with the latest attempt at a data collection for m-mode analysis.  I recommend a single sub-band data collection be scheduled for 24 hours (switching of the slow vis buffer during this period), followed by data transfer out.

yupinghuang, 2023-08-20:
Here are two fun ideas. Try having some fun with zfs set primarycache or
use part of the nvme as cache for directory metadata.

^thats a note to self.

yupinghuang, 2023-08-30:
So it seems like disabling zfs's prefetching gives a more stable read latency and makes the data recorder happier.

I was able to run the data recorder plus an rsync plus a `du -sh .` on the `slow` directory without issue after disabling prefetch, whereas on a different node where prefetch wasn't disabled the data recorder started to fall behind fairly quickly.

So I'm running
```echo 1 >> /sys/module/zfs/parameters/zfs_prefetch_disable```

on all nodes and also adding `/etc/modprobe.d/zfs.conf` to all nodes with

```
options zfs zfs_prefetch_disable=1
```

caseyjlaw, 2023-09-07:
@rh-codebase to try "renice" data recorder priority on calim nodes 04 07 08.

yupinghuang, 2023-09-07:
Oh I already tried that

yupinghuang, 2023-09-07:
Actually I'm gonna close this issue. `rsync` hasn't caused the data recorder since the `spl_taskq_thread_priority` change. The remaining issue is that data recorder starts to lag behind because of quota.

closed 2023-09-07 by yupinghuang