ovro-lwa/lwa-issues #462: Resource Exhausted error

iveydavis, 2023-08-17:
I think I did an accidental stress test by using a high tracking cadence for the beamformer. I saw that the tracking loop was reporting "failed to command lxdlwagpuXX (pipeline NN)" so I stopped the tracking. Interestingly, it looks like the data recorder didn't have any issues with finishing the observation. 

I was hoping that resetting the lwa-xeng-etcd.service and bifrost_etcd_bridge service would fix things, but it doesn't seem to.

```
con.configure_xengine(full=True)
2023-08-17 05:57:48,181 - mnc.control - INFO - Stopping/starting pipelines with 20s sleep
---------------------------------------------------------------------_InactiveRpcError                   Traceback (most recent call last)<ipython-input-9-74f464e02f3c> in <module>()
----> 1 con.configure_xengine(full=True)

/home/pipeline/proj/lwa-shell/mnc_python/mnc/control.py in configure_xengine(self, recorders, calibratebeams, full, timeout)
    234             logger.info("Stopping/starting pipelines with 20s sleep")
    235             # stop before starting
--> 236             self.pcontroller.stop_pipelines()
    237             self.stop_xengine()
    238             self.start_xengine(timeout=timeout)

/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/lwa352_pipeline_control/lwa352_pipeline_control.py in stop_pipelines(self)    155         """
    156         for pl in self.pipelines:
--> 157             pl.stop_pipeline(force=True) #Will kill all pipelines on the remote servers
    158         time.sleep(10)
    159         stopped = True

/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/lwa352_pipeline_control/lwa352_pipeline_control.py in stop_pipeline(self, force)
    390         :type force: bool
    391         """
--> 392         self.corr_interface.send_command(self.host, cmd='stop_pipeline', block='xctrl', xid=self.pipeline_id, force=force)
    393
    394     def check_connection(self, timeout=1):

/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/lwa352_pipeline_control/etcd_control.py in send_command(self, host, pipeline, block, inst_id, cmd, timeout, **kwargs)
    299         watch_id = self.ec.add_watch_callback(resp_key, response_callback)
    300         # send command
--> 301         self.ec.put(cmd_key, command_json)
    302         starttime = time.time()
    303         while(True):

/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/etcd3/client.py in handler(*args, **kwargs)
     46                 return f(*args, **kwargs)
     47             except grpc.RpcError as exc:
---> 48                 _translate_exception(exc)
     49
     50     return functools.wraps(f)(handler)

/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/etcd3/client.py in handler(*args, **kwargs)
     44         def handler(*args, **kwargs):
     45             try:
---> 46                 return f(*args, **kwargs)
     47             except grpc.RpcError as exc:
     48                 _translate_exception(exc)

/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/etcd3/client.py in put(self, key, value, lease, prev_kv)
    422             self.timeout,
    423             credentials=self.call_credentials,
--> 424             metadata=self.metadata
    425         )
    426

/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/grpc/_channel.py in __call__(self, request, timeout, metadata, credentials, wait_for_ready, compression)
    944         state, call, = self._blocking(request, timeout, metadata, credentials,
    945                                       wait_for_ready, compression)
--> 946         return _end_unary_response_blocking(state, call, False, None)
    947
    948     def with_call(self,

/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/grpc/_channel.py in _end_unary_response_blocking(state, call, with_call, deadline)
    847             return state.response
    848     else:
--> 849         raise _InactiveRpcError(state)
    850
    851

_InactiveRpcError: <_InactiveRpcError of RPC that terminated with:
        status = StatusCode.RESOURCE_EXHAUSTED
        details = "etcdserver: mvcc: database space exceeded"
        debug_error_string = "{"created":"@1692251868.185058533","description":"Error received from peer ipv4:10.40.0.87:2379","file":"src/core/lib/surface/call.cc","file_line":966,"grpc_message":"etcdserver: mvcc: database space exceeded","grpc_status":8}"
```

iveydavis, 2023-08-17:
I had three beams going, one that at the usual 15s tracking update cadence and two that were at a 1s cadence.

in one terminal:
```
con = control.Controllercontrol.Controller('/home/pipeline/proj/lwa-shell/mnc_python/config/lwa_config_calim_ivey.yaml')
con.configure_xengine(['dr6'],full = False, calibratebeams = True)
con.conf['xengines']['update_interval'] = 1
con.control_beamformer(6, targetname = 'EV Lac', track = True)
```
In another terminal:
```
con = control.Controllercontrol.Controller('/home/pipeline/proj/lwa-shell/mnc_python/config/lwa_config_calim_ivey.yaml')
con.configure_xengine(['dr7'],full = False, calibratebeams = True)
con.conf['xengines']['update_interval'] = 1
con.control_beamformer(7, targetname = 'Cyg A', track = True)
```
And in the last terminal
```
con = control.Controllercontrol.Controller('/opt/devel/dgary/lwa_config_calim.yaml')
con.configure_xengine(['dr5'],full = False, calibratebeams = True)
con.control_beamformer(5, targetname = 'Cyg A', track = True)
```

caseyjlaw, 2023-08-17:
I'm not sure if the beam control did this, but we are certainly down.

@rh-codebase The etc database has run out of disk space. Array operations are down since about 12 hours ago. Can you fix it?

rh-codebase, 2023-08-17:
etcd database was cleared and process restarted. Wx data are flowing through the system into Grafana so that part is now working again. try restarting services.

caseyjlaw, 2023-08-17:
Thanks, Rick. I'll take a crack at this.
Do we know how to anticipate this database filling? I can set a grafana alert on etcd info, if I know the right monitor point and value.

rh-codebase, 2023-08-17:
looking into that as well as seeing recommendations for defrag-ing the database but that looks to need to have etcd stopped so will have to think about it. the clients need to be resilient against etcd going offline and coming back. Also noticed that etcd was running with the default 2GB database size. I thought that was bumped to 4GB. Restarted with 4GB.  Grafana does have a graph on database size, so will look to see what monitor point it's using. 3.5 was announced so will get that in when I find time to build it.

rh-codebase, 2023-08-17:
yeah. can't figure out why. working on a newer version with better logging.

caseyjlaw, 2023-08-17:
I ran Ivey's fast-update beamformer observation again. Everything seems fine, so I think it was a coincidence with this disk space issue.
I'll close this.

rh-codebase, 2023-08-17:
Well, it might have caused the db to exceed if the db was close to exceeding. We need to see if 2GB is what we expect to be in the db so all the subsystems can send us their estimates of how much they are pushing into etcd. Maybe a timing issue with db compaction and ingest rate. Let's keep open until we know more and see what we should be setting the db  size to.

caseyjlaw, 2023-12-14:
This seems to be related to filled disk.

closed 2023-12-14 by caseyjlaw