ovro-lwa/lwa-issues #464: Dropped/Zeroed subbands in the slow visibility data

isella, 2023-08-25:
Subbands are (randomly) dropped (i.e., not written) or zeroed out in slow visibility ms files. 

In the first case --- when subbands visibility files are not written --- the system health dashboard reports errors for the corresponding data recorder. 

In the second case --- when visibility files are written but all values are set to 0 --- no error in the data recorder status are reported. 
ms files with 0ed visibility also show a mismatch between the time stamp reported in the file name and the data time stamp reported in the ms file. 

On Aug 24, Jack implemented a change to the x-engine that may synchronize the two subbands processed on each GPU. However, as of this morning (Aug 25), there are still subbands with zero values.

jaycedowell, 2023-08-29:
I think what is happening here is competition between the data recorders and people using calim01-08 for data reduction.  The scenario is something like:
 1. Everything is working as expected for the data recorders and packet loss is ~0%.
 2. Someone logs onto calim01-08 and starts reducing data.
 3. The data recorders on that machine become resource limited and start dropping packets.
 4. Eventually the data recorders fall so far behind on packet capture that they are catching nothing.  The packet loss rate goes to 100% and zeros are written in place of the missing packets[^1].

Once you reach (4) the only way to recover is to restart the affected DR services.  This condition should be noticeable on the dashboard, though, since the any packet loss over 10% triggers an error condition.  Occasional slowdowns, i.e., situations where you reach (3) but it isn't bad enough to get to (4), will lead to error conditions but they may be transient since errors are automatically cleared when/if things get better.

I have mostly anecdotal evidence that supports this from watching things over the last few days:
 1. calim07 is running at a high load (load average of 30-40) at the times I have checked on it.  I also see processing occasionally entering the D state on `top` which is suggestive of a high I/O load.
 2. One of the slow vis. recorders on calim04 reported a write time for a measurement set of ~1 min.  This is ~12x longer than it typically should take.

[^1]: The X-engines are still sending packets but because the data recorder is lagging it ends up looking for packets that potentially arrived minutes in the past.

jaycedowell, 2023-08-29:
> ms files with 0ed visibility also show a mismatch between the time stamp reported in the file name and the data time stamp reported in the ms file.

I can't see how this could happen looking at how the data recorders deal with time.  Can someone point me to a measurement set that shows this?

yupinghuang, 2023-08-29:
I'm tempted to say that packet loss issue is all I/O, because even on calim07/04 when load average is sky high (by accident sometimes), the drs do not report loss unless recording is turned on. It makes sense that zfs needs cores to do its thing (or maybe it's busy shrinking its 250 GB cache...) The slow vis recorder really shouldn't be that resource hungry.

I tried fiddling with zfs settings and so far no dice. Maybe I should try shrinking the cache drastically.

Here are two quick things once can try...
- switch to a multiprocessing pool so that I/O doesn't hold up packet capture
- turn on the tar thing once again (?)

yupinghuang, 2023-08-29:
And the timestamp issue seems to be https://github.com/ovro-lwa/lwa-issues/issues/225 again

jaycedowell, 2023-08-29:
I think that is consistent with reaching (4).  The timestamps are based on the packets coming in so a lagging DR leads to lagging MS files.

rh-codebase, 2023-08-29:
As much as I love ZFS, perhaps using XFS is worth a try? Jack found on the GPU nodes that XFS on top of mdadm was more performant than the ZFS setup I implemented.

jaycedowell, 2023-08-29:
> switch to a multiprocessing pool so that I/O doesn't hold up packet capture

Maybe?  I'm concerned about memory usage if there is a prolonged slow down on writing to disk.  Plus #445 is still a thing.

> turn on the tar thing once again (?)

I was going to say that this would be easy but I think I've removed the "create the measurement set in /dev/shm first" part of the code.  It could be added back in.

@rh-codebase also suggested trying to change the priority of the recorders to see if that would help.  We could move the services over to system services or maybe modify the limits for pipeline so that it we could stick with a user service.

jack-h, 2023-08-31:
@rh-codebase just to clarify -- I didn't so much fund that XFS was faster than ZFS, but my code relied on the directio file writing option for speed, and this didn't play nicely with zfs.

yupinghuang, 2023-09-04:
so I turned off pre-fetching and set `spl_taskq_thread_priority` to 0 so that zfs threads have higher priority. That seems to have help...?

did that a few days ago on calim04 and that seemed to have run alright with an rsync and some gregg stuff.

I did that on all other nodes and started the data recorder a few hours ago. Ran the following on calim01 for half an hour (along with a `du -sh .` which took a while...) and the data recorder seemed alright. Although issues might arise as the data recorder runs on...

```
[yuping@lwacalim01 ~]$ sudo stress --cpu 20 -v --vm 8 --vm-bytes 40G
```

that brings

```
(base) [yuping@lwacalim01 ~]$ free -g
              total        used        free      shared  buff/cache   available
Mem:            503         343         158           0           1         156
Swap:             3           0           3
(base) [yuping@lwacalim01 ~]$ w
 06:16:42 up  5:00,  3 users,  load average: 29.15, 30.47, 23.53
USER     TTY      FROM             LOGIN@   IDLE   JCPU   PCPU WHAT
yuping   pts/0    10.40.0.4        05:57   22.00s 10:14   0.02s sshd: yuping [priv]                                                                                                                              
yuping   pts/1    10.40.0.4        05:58   12:02   3.82s  3.74s du -sh .
yuping   pts/2    10.40.0.4        06:05    0.00s  0.07s  0.02s w
```

will hammer it a bit harder tmr....

yupinghuang, 2023-09-04:
ok now it's doing `Packet loss during receive >10% (24.8% missing)` on and off when it's time to calculate quota and removing files. But at least it didn't get to 100%...

yupinghuang, 2023-09-07:
so...it seems like after a while...quota gets slower and slower and eventually it holds  the GIL for too long and the capture thread falls behind. Sigh
```
2023-09-06 05:33:24 [INFO    ] MSStart - Scheduled recording start for 2023-09-06 05:33:39.910925 to 2024-09-05 05:33:39.910925 to /data01/slow/
2023-09-06 05:34:17 [INFO    ] Started operation - <MeasurementSetWriter filename='/data01/slow', start_time='2023-09-06
    05:33:39.910925', stop_time='2024-09-05 05:33:39.910925',
    reduction=None>
2023-09-06 06:26:34 [INFO    ] StorageLogger: Updating storage usage in /data01/slow.
2023-09-06 06:28:35 [INFO    ] Quota: Number of items to remove: 626
2023-09-06 07:26:35 [INFO    ] StorageLogger: Updating storage usage in /data01/slow.
2023-09-06 07:27:03 [INFO    ] Quota: Number of items to remove: 718
2023-09-06 08:26:35 [INFO    ] StorageLogger: Updating storage usage in /data01/slow.
2023-09-06 08:27:38 [INFO    ] Quota: Number of items to remove: 718
2023-09-06 09:26:35 [INFO    ] StorageLogger: Updating storage usage in /data01/slow.
2023-09-06 09:28:57 [INFO    ] Quota: Number of items to remove: 718
2023-09-06 10:26:35 [INFO    ] StorageLogger: Updating storage usage in /data01/slow.
2023-09-06 10:28:00 [INFO    ] Quota: Number of items to remove: 718
2023-09-06 11:26:35 [INFO    ] StorageLogger: Updating storage usage in /data01/slow.
2023-09-06 11:28:51 [INFO    ] Quota: Number of items to remove: 718
2023-09-06 12:26:36 [INFO    ] StorageLogger: Updating storage usage in /data01/slow.
2023-09-06 12:27:31 [INFO    ] Quota: Number of items to remove: 718
2023-09-06 13:26:37 [INFO    ] StorageLogger: Updating storage usage in /data01/slow.
2023-09-06 13:27:14 [INFO    ] Quota: Number of items to remove: 718
2023-09-06 14:26:39 [INFO    ] StorageLogger: Updating storage usage in /data01/slow.
2023-09-06 14:28:05 [INFO    ] Quota: Number of items to remove: 718
2023-09-06 15:26:40 [INFO    ] StorageLogger: Updating storage usage in /data01/slow.
2023-09-06 15:28:24 [INFO    ] Quota: Number of items to remove: 718
2023-09-06 16:26:40 [INFO    ] StorageLogger: Updating storage usage in /data01/slow.
2023-09-06 16:28:06 [INFO    ] Quota: Number of items to remove: 718
2023-09-06 17:26:41 [INFO    ] StorageLogger: Updating storage usage in /data01/slow.
2023-09-06 17:27:06 [INFO    ] Quota: Number of items to remove: 718
2023-09-06 18:26:43 [INFO    ] StorageLogger: Updating storage usage in /data01/slow.
2023-09-06 18:26:57 [INFO    ] Quota: Number of items to remove: 290
2023-09-06 18:37:26 [INFO    ] Current pipeline lag is 0:00:30.649095
2023-09-06 19:26:43 [INFO    ] StorageLogger: Updating storage usage in /data01/slow.
2023-09-06 19:27:56 [INFO    ] Quota: Number of items to remove: 672
2023-09-06 20:26:43 [INFO    ] StorageLogger: Updating storage usage in /data01/slow.
2023-09-06 20:27:56 [INFO    ] Quota: Number of items to remove: 718
2023-09-06 21:26:43 [INFO    ] StorageLogger: Updating storage usage in /data01/slow.
2023-09-06 21:42:10 [INFO    ] Quota: Number of items to remove: 718
2023-09-06 22:26:43 [INFO    ] StorageLogger: Updating storage usage in /data01/slow.
2023-09-06 22:28:57 [INFO    ] Quota: Number of items to remove: 718
2023-09-06 23:26:43 [INFO    ] StorageLogger: Updating storage usage in /data01/slow.
2023-09-06 23:28:08 [INFO    ] Quota: Number of items to remove: 718
2023-09-07 00:26:43 [INFO    ] StorageLogger: Updating storage usage in /data01/slow.
2023-09-07 01:08:03 [INFO    ] Quota: Number of items to remove: 716
2023-09-07 01:26:43 [INFO    ] StorageLogger: Updating storage usage in /data01/slow.
2023-09-07 01:47:16 [WARNING ] Received signal 15 SIGTERM
```

yupinghuang, 2023-09-27:
Gonna declare partial victory. calim05 has been running a full cpu load for the past few days and things seemed to be fine. We have only been seeing transient data loss, as long as someone restarts the dr services every 48 hours or so. So gonna close this in favor of feature requests https://github.com/lwa-project/ovro_data_recorder/issues/37 and https://github.com/lwa-project/ovro_data_recorder/issues/36