ovro-lwa/lwa-issues #465: Missing files after triggering & x-engine pipelines in waiting state

nkosogor, 2023-08-30:
Some inconsistencies in the number of files on different GPU servers (each should have 44 files):
- gpu2:data0 has 0 files (files missing).
- gpu3:data0 has only 6 files.
- gpu5:data1 has only 6 files.

After the trigger, the x-engine pipelines have entered a ‘waiting’ state. On the dashboard, only the triggered pipelines are set to the waiting state (gpu01-2 and 3; gpu02-2 and 3...).

jack-h, 2023-08-31:
Logs say:

```
2023-08-29 16:30:05 [INFO    ] Sending beam 2 to 10.41.0.98:20001
2023-08-29 16:30:06 [INFO    ] Sending beam 2 to 10.41.0.77:20001
2023-08-29 19:12:25 [INFO    ] Sending beam 2 to 10.41.0.98:20001
2023-08-29 19:12:26 [INFO    ] Sending beam 2 to 10.41.0.77:20001
2023-08-30 04:24:17 [INFO    ] TRIGGERED DUMP >> Opening /data0/lwa-dump-1693369356.65.tbf.0
2023-08-30 04:24:17 [INFO    ] Start seq is 40515180735360
2023-08-30 04:24:17 [INFO    ] CORR >> new input sequence!
2023-08-30 04:24:19 [INFO    ] CORR >> New start time set to 40504878677760. Accumulating 2400 samples
```

Triggered data (eg on GPU07) were 4 files with one timestamp, then 39 with another. My theory is that after the trigger fired, packet loss caused a bifrost sequence reset. As a result, the buffer dumper wrote it's ~256 GB buffer (which is about 4 files worth) and then was hit with a new sequence ID, so wrote a bunch of new files with that ID.

Fundamentally, I think the issue here is that triggers seem to be resulting in enough packet loss that the bifrost data sequence relocks to a new timestamp. Amongst other things, this causes the correlator to reset - this issue, at least, could probably be solved by having the correlator relock to an accumulation boundary which matches the boundaries it was previously using.

jack-h, 2023-09-06:
I..... can't recreate this. I fired up the correlator, running slow visibility output, and triggering short (~0.3GB) medium (~3GB) or long (3x64GB) files doesn't seem to have any adverse effects.
@nkosogor do you think you can come up with a way to reliably recreate this? Perhaps it has something to do with where pipelines are writing files, or what the beamformers are doing, or..... something else(?).

My test (with 64GB files) was simply this:
```
from mnc.control import Controller

con = Controller()
con.configure_xengine(['drvs'],full=True)
for p in con.pipelines:
     if p.pipeline_id < 2:
         continue
     p.triggered_dump.trigger(ntime_per_file=2*480000, nfile=3, dump_path='/data%d' % (p.pipeline_id % 2))
```

On multiple runs there was no extra packet loss and the X-engines continued to run happily

nkosogor, 2023-11-15:
@jack-h, I thought the issue had been resolved, as I was able to successfully trigger a few times in the past two months, and the x-engines were working fine. 
However, today while Casey and I were conducting some tests the x-engines entered a ‘waiting’ state with `curr_sample=NaN` after I triggered.

Below is the code I used (the same code I've used in the past):
```
from mnc import control
con = control.Controller()
pipelines = [p for p in con.pipelines if p.pipeline_id in [2, 3]]

ntime_per_file = 1000000
nfile = 7
path_map = {2: '/data0/', 3: '/data1/'}

for pipeline in pipelines:
    if pipeline.pipeline_id in path_map:
        path = path_map[pipeline.pipeline_id]
        pipeline.triggered_dump.trigger(ntime_per_file=ntime_per_file, nfile=nfile, dump_path=path)

```