ovro-lwa/lwa-issues #466: Need to set rules for cal table directories

iveydavis, 2023-09-07:
More specifically, need to set standards for what files are kept in the `/home/pipeline/caltables/latest/` directory.

caseyjlaw, 2023-09-08:
Thanks, Ivey. I think the cal tables are large enough that it makes sense to keep them on disk. We just need rules about how to make and use them. Below is my proposal; if I tagged you, please comment.

Some requirements:
- We will use CASA calibration tables written to disk on cal-im. (too big for etcd)
- Need a caltable that is applied by default that will work for most users
- Need to allow users to generate custom tables that they can specify when needed
- Caltables should have some description of how they were made (e.g., core only, what algorithm and input data)

Questions in my mind:
- How useful is it to keep older caltables?
- What is best for default?

Proposed scheme:
- `/home/pipeline/caltables/latest` is a symlink that points to a directory updated by Yuping's script
- directory names have a convention "yyyy-mm-dd_<tag>", where <tag> can be "default" or "core" or whatever.
- script will also output some metadata to describe how caltable was made (json with some standard fields, hopefully)
- anyone logged in as pipeline user can create a new directory with this name convention and a <tag> of their choice, but "default" means that it was made by Yuping's script.

dgary50, 2023-09-08:
>    - directory names have a convention "yyyy-mm-dd_", where can be
>    this name convention and a of their choice, but "default" means that it was

One thing I am curious about: is it really necessary to have cal tables for
different sets of antennas?  Why can't we have one set of cal tables (good
for all currently in use antennas) and then use a weighting scheme or
good/bad ants selection file that selects which antennas are used at the
application stage?  For example, a parameter that specifies a badants file,
or a string like "all" or "core" to choose two likely favorite options, in
the configure_xengine() call?

Regards,
Dale

yupinghuang, 2023-09-08:
> 

Wouldn't that be great. Oh wait it's already there! https://github.com/ovro-lwa/mnc_python/blob/main/mnc/control.py#L302

In all seriousness though, I'd like someone else to take this on. I don't use the beamformer.

iveydavis, 2023-09-08:
@dgary50 which antennas are used aren't my concern so much (as yuping pointed out, there's already control in the mnc now for specifying which antennas you want to use, although I suppose it's up to you whether you feel cal files should be made with the antennas flagged rather than have them made under the assumption the antennas will be used). A bigger concern is that the caltables I'll be using presumably aren't re-usable from night to night because of the inclusion of the outriggers. I've been making them around 7-10pm (moving forward, I'll have to make them even earlier so I can actually co-observe with Flarescope), but you want caltables based on data collected later in the night, right?

caseyjlaw, 2023-09-28:
Need to associate caltables with settings configuration.
Will migrate toward full-array calibration and core-only selection done in beamformer. So no need to define core-only locally with caltables.

caseyjlaw, 2023-09-28:
@yupinghuang `mnc-python` has a bit of code to understand the state of the arx/f-engine settings. Could you use this to write a text file with the settings filename in the caltable directory?
```
from mnc import settings
ss = settings.Settings()
filename = ss.get_last_settings()
```

Alternatively, you can just set it to the latest file in advance of running calibration observations:
```
ss.load_feng()
ss.load_arx()
ss.update_log()
```

closed 2023-10-19 by caseyjlaw