ovro-lwa/lwa-issues #467: Proposal for antenna flagging

caseyjlaw, 2023-09-07:
Related to #234 #437.

We have multiple ways of identifying bad antennas and multiple users of bad antenna information. It would help to use etcd to store bad antenna information so:
1. We have a flexible way of adding bad antenna identification algorithms
2. Anyone can read it
3. We can track it in time with influx/grafana

My proposal:
- Create a key hierarchy `/mon/anthealth` to share antenna health info
- Each source of ant health info can make a key under it, e.g., `/mon/anthealth/selfcorr/<n>` to hold a flag for antenna n based on a selfcorr algorithm. This is extensible, as new keys can be added later, as we invent them.
- Example payload `{'time': mjd, 'ant_num': ant_num, 'flagged': True/False}`. Could also store non-boolean values, if user wants to threshold on it.
- A special name `/mon/anthealth/<n>` holds union of all (or "best") flags. E.g., `{'time': nnn, 'ant_num: ant_num, 'flagged': True/False, 'selfcorr': True/False, etc...}`
- The union of all flags could be updated by a watch function on the lower-level flags.

Python interface example:
```
> from mnc import anthealth
> [calculate array of bad antennas]
> anthealth.set_badants('selfcorr', badantlist)
> badants = anthealth.get_badants('selfcorr')
> print(badants)
[103, 205, 206, ...]
```

yupinghuang, 2023-09-08:
just gonna be real annoying and ask what is an `ant_num`.....I assume it's the correlator number.

I'd push the union of flags to the `mnc.anthealth` code. Otherwise you need to deal with locking etc in etcd which will not be fun.

In the payload scheme...how does the algorithm suggest that an antenna is good again after it flags it? Does it need to look through all the antennas each time it submit a flag to make sure that antennas are unflagged? That sounds like a lot of work.

`/mon/anthealth/selfcorr/` with {'time': time, 'flagged': [True, True, True]} with flagged being a dimension 352 boolean array may be a saner choice.

---------------------------------------------------------------------
The time axis is kinda important, how is this reflected in the code and k/v design? Ideally, you want to avoid going through all the flags that there was to find out what the flags are 23 hours ago.

As a science user, the question I'd most likely to ask is "which antennas are bad at time x?" and I want to to that pretty often.

For tracking instrument health, you'd want to ask "what did this antenna look like for the last month" with a higher latency tolerated.

My understanding is that these keys will be ingested into influx. But I don't know if they are done whenever new flags are pushed or on a regular cadence.

caseyjlaw, 2023-09-28:
Larry asks: Can we preserve individual signals (pols) that are good, even if one signal on an antenna is bad?

caseyjlaw, 2023-10-05:
Ideas for generating bad antenna lists:
- selfcorr (via andrea)
- referencing spectra to historical (median?) example (via marin)
- newcal drop-out analysis (via ruby)

closed 2023-10-19 by caseyjlaw