ovro-lwa/lwa-issues #475: Calibrated beamformer observation with two antennas to measure backend bf scaling

caseyjlaw, 2023-09-21:


jack-h, 2023-09-28:
If you want known test vectors -- From page 39 of the F-engine manual -- https://github.com/realtimeradio/caltech-lwa/blob/master/docs/build/latex/LWA352SNAP2F-Engine.pdf

[image]

I suggests `write_freq_ramp()` and `tvg_enable()`

Test vector interpretation is detailed here -- https://github.com/ovro-lwa/mnc_python/pull/10#issuecomment-1434377728

surajit4444mondal, 2023-09-28:
I first did a bandpass calibration in CASA and then applied to the dataset. The dataset was then phase-rotated to the location of CygA. Next I use the following code to simulate the beam (this is now what I think matches the beamformer implementation available now)

`from casatools import table
import numpy as np
import matplotlib.pyplot as plt

def get_baseline_index(ant_id1, ant_id2,num_ant=352,auto_corr=True):
    '''
    ant_id1 <=ant_id2
    '''
    if auto_corr==True:
        return num_ant*ant_id1+ant_id2-ant_id1*(ant_id1+1)/2
    else:
        return num_ant*ant_id1+ant_id2-(ant_id1+1)*(ant_id1+2)/2

vis='20230707_102001_55MHz_cygA_phased.ms'
num_ant=352
time_normalisation=24

tb=table()
try:
    tb.open(vis)
    data=tb.getcol('CORRECTED_DATA')
    flag=tb.getcol('FLAG')
finally:
    tb.close()
    
shape=data.shape
num_pol=shape[0]
num_chan=shape[1]

auto_corr_inds=np.array(get_baseline_index(np.arange(0,352),np.arange(0,352)),dtype=int)

flag[:,:,auto_corr_inds]=False
        
pos=np.where(flag==True)
data[pos]=np.nan

phased_flux=(2*np.nansum(np.real(data),axis=2)-np.nansum(data[:,:,auto_corr_inds],axis=2))*time_normalisation #/(num_ant**2)

plt.plot(np.abs(phased_flux[0,:]));plt.show()
`
Next if we define phased flux as 
`phased_flux=(2*np.nansum(np.real(data),axis=2)-np.nansum(data[:,:,auto_corr_inds],axis=2))/(num_ant**2)`

and plot that, we will see the number which matches the CygA flux seen in image (before primary beam correction was done)

caseyjlaw, 2023-10-19:
We have an understanding of beamformer scaling now and would like to make changes to remove the dependence on antenna selection. Being tracked in #382, so closing this.

closed 2023-10-19 by caseyjlaw