ovro-lwa/lwa-issues #476: SNAPs become unresponsive to tapcp after many days running

jack-h, 2023-09-23:
On two occurrences, all SNAP boards have become unresponsive to TFTP commands. Pings continue to work.

The second occurrence happened ~18 days after the first.
Logs reveal the first failed TFTP transaction as:

```
(base) [pipeline@lwacalim10 ~]$ for f in `ls ~/logs/lwa-feng-etcdsnap*.out`; do grep "2023-09-23" $f -A 40 | grep -i "Access Violation" -m 1 -B 30 | grep ERROR; done
2023-09-23 06:38:48,887 - /opt/devel/pipeline/envs/deployment/bin/lwa_start_feng_etcd_service.py:snap01 - ERROR - Error polling stats
2023-09-23 01:05:58,791 - /opt/devel/pipeline/envs/deployment/bin/lwa_start_feng_etcd_service.py:snap03 - ERROR - Error polling stats
2023-09-23 04:27:07,701 - /opt/devel/pipeline/envs/deployment/bin/lwa_start_feng_etcd_service.py:snap04 - ERROR - Error polling stats
2023-09-23 01:34:22,252 - /opt/devel/pipeline/envs/deployment/bin/lwa_start_feng_etcd_service.py:snap05 - ERROR - Error polling stats
2023-09-23 02:41:47,989 - /opt/devel/pipeline/envs/deployment/bin/lwa_start_feng_etcd_service.py:snap06 - ERROR - Error polling stats
2023-09-23 05:22:45,867 - /opt/devel/pipeline/envs/deployment/bin/lwa_start_feng_etcd_service.py:snap07 - ERROR - Error polling stats
2023-09-23 01:32:38,665 - /opt/devel/pipeline/envs/deployment/bin/lwa_start_feng_etcd_service.py:snap08 - ERROR - Error polling stats
2023-09-23 02:38:27,787 - /opt/devel/pipeline/envs/deployment/bin/lwa_start_feng_etcd_service.py:snap09 - ERROR - Error polling stats
2023-09-23 09:03:20,201 - /opt/devel/pipeline/envs/deployment/bin/lwa_start_feng_etcd_service.py:snap10 - ERROR - Error polling stats
2023-09-23 07:47:25,149 - /opt/devel/pipeline/envs/deployment/bin/lwa_start_feng_etcd_service.py:snap11 - ERROR - Error polling stats
```

I.e., all boards fail on the same day, but at very different times.

I had initially thought the error might be due to a memory leak causing the microblaze address table to become corrupted. But, TFTP `get` of `/help` (which isn't in the `core_info.tab` map) also doesn't work.
All commands return `"Only one connection at a time is supported"`. This probably means the error comes from inside the TFTP implementation -- https://github.com/realtimeradio/mlib_devel/blob/b590fd81aa08293ed8c0ee9d61ebb889752874fa/jasper_library/sw/jam/lwip/src/apps/tftp/tftp_server.c#L212

yupinghuang, 2023-10-11:
Here's full stacktrace and an error code...

```
Oct  8 09:30:33 lwacalim10 journal[2215091]: 2023-10-08 09:30:33,645 - /opt/devel/pipeline/envs/deployment/bin/lwa_start_feng_etcd_service.py:snap11 - ERROR - Error polling stats#012Traceback (most recent call last):#012  File "/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/lwa_f/snap2_feng_etcd_client.py", line 622, in poll_stats#012    stats, flags = self.feng.get_status_all()#012  File "/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/lwa_f/snap2_fengine.py", line 178, in get_status_all#012    if not self.blocks['fpga'].is_programmed():#012  File "/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/lwa_f/blocks/fpga.py", line 105, in is_programmed#012    return 'version_version' in self.listdev()#012  File "/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/lwa_f/blocks/block.py", line 155, in listdev#012    devs = self.host.listdev()#012  File "/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/casperfpga/casperfpga.py", line 287, in listdev#012    return self.transport.listdev()#012  File "/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/casperfpga/transport_tapcp.py", line 171, in listdev#012    self.t.download('/listdev'.encode(), buf, timeout=self.timeout)#012  File "/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/tftpy-0.8.0-py3.6.egg/tftpy/TftpClient.py", line 58, in download#012    self.context.start()#012  File "/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/tftpy-0.8.0-py3.6.egg/tftpy/TftpContexts.py", line 402, in start#012    self.cycle()#012  File "/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/tftpy-0.8.0-py3.6.egg/tftpy/TftpContexts.py", line 202, in cycle#012    self.state = self.state.handle(recvpkt, raddress, rport)#012  File "/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/tftpy-0.8.0-py3.6.egg/tftpy/TftpStates.py", line 490, in handle#012    raise TftpException("Received ERR from peer: " + str(pkt))#012tftpy.TftpShared.TftpException: Received ERR from peer: ERR packet: errorcode = 2#012    msg = b'Access violation'
```

```
Oct  9 10:30:27 lwacalim10 journal[2214578]: 2023-10-09 10:30:27,994 - /opt/devel/pipeline/envs/deployment/bin/lwa_start_feng_etcd_service.py:snap03 - ERROR - Error polling stats#012Traceback (most recent call last):#012  File "/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/lwa_f/snap2_feng_etcd_client.py", line 622, in poll_stats#012    stats, flags = self.feng.get_status_all()#012  File "/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/lwa_f/snap2_fengine.py", line 178, in get_status_all#012    if not self.blocks['fpga'].is_programmed():#012  File "/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/lwa_f/blocks/fpga.py", line 105, in is_programmed#012    return 'version_version' in self.listdev()#012  File "/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/lwa_f/blocks/block.py", line 155, in listdev#012    devs = self.host.listdev()#012  File "/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/casperfpga/casperfpga.py", line 287, in listdev#012    return self.transport.listdev()#012  File "/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/casperfpga/transport_tapcp.py", line 171, in listdev#012    self.t.download('/listdev'.encode(), buf, timeout=self.timeout)#012  File "/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/tftpy-0.8.0-py3.6.egg/tftpy/TftpClient.py", line 58, in download#012    self.context.start()#012  File "/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/tftpy-0.8.0-py3.6.egg/tftpy/TftpContexts.py", line 402, in start#012    self.cycle()#012  File "/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/tftpy-0.8.0-py3.6.egg/tftpy/TftpContexts.py", line 202, in cycle#012    self.state = self.state.handle(recvpkt, raddress, rport)#012  File "/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/tftpy-0.8.0-py3.6.egg/tftpy/TftpStates.py", line 490, in handle#012    raise TftpException("Received ERR from peer: " + str(pkt))#012tftpy.TftpShared.TftpException: Received ERR from peer: ERR packet: errorcode = 2#012    msg = b'Access violation'
```

caseyjlaw, 2023-10-30:
I'm running into issues when trying to recover from this failure mode. Likely stuff that Yuping already noted, but we never fixed in code or docs.

Steps I've taken:
- went to pi@lwapowersnap and did `./powersnap.sh 0 toggle` and `./powersnap.sh 1 toggle`.
- went to pipeline@lwacalim10 and restarted feng etcd services.
- tried mnc-python `start_fengine`, but got lots of `Received 0 of 11 responses`.
- tried using `lwa_observe_etcd.py` and am still seeing those responses.

I'm seeing this error in the fengine logs after the last attempt:
```
lwa_f.snap2_fengine:snap01 - ERROR - Failed to read and decode .fpg header from flash
```

What am I missing?

jack-h, 2023-11-17:
>    - went to ***@***.*** and did ./powersnap.sh 0 toggle and ./powersnap.sh
>    - went to ***@***.*** and restarted feng etcd services.

Failed to decode .fpg header from flash shouldn't happen, but indicates
that the image in flash is corrupt. Reuploading a bitstream should fix
things.
It would be nice if I understood why the image in flash is becoming corrupt
on power cycling, but I don't. That said, if there is some memory leak in
the uBlaze code, presumably it could be doing absolutely anything, up to
and including erroneously writing to flash.