ovro-lwa/lwa-issues #478: calim runs close to 90C on full cpu load

yupinghuang, 2023-09-25:
This happens when the load is high (this is calm05, load was 52 or something, number of cores is 48) and can run for arbitrary amount of time as long as the load is high.

A bit too high for my taste for a sustained amount of time. @rh-codebase thoughts/tips/tricks?

[image]

yupinghuang, 2023-09-25:
is the ~15min cycle the AC????

I set up slack integration for grafana and AN alert on calim05's temperature > 85C. The alert turned out to be a huge chore since grafana doesn't support template variables for alerts...

Is there a config file or something that can make setting up alerts easier than manually doing it n_nodes times...?

yupinghuang, 2023-09-28:
Me to investigate the BIOS and manual to see if 90C is cause for concern.

@Princemongo to check if the fan on the calim rack is working well.

yupinghuang, 2023-09-28:
On the calimipmi, the critical temperature is set to 100C. So 90C sustained is not good.

coincidentally, there's a configuration for fan speed on ipmi. So I'll try that out and report back

yupinghuang, 2023-09-28:
[image]

Poke around on ipmi. At around 18:23 I switched the fan mode to "full".  And then at 18:37 I set it back to "optimal", which was the default. At 90C, under "optimal" mode, the CPU fans are only running at 85% the full speed, which is a little puzzling.

The 20C variation at 10-15min timescale also, maybe, doesn't make a ton of sense?

I think it's still worth it to check the rack fan in the shelter. And I'll see if I can get the motherboard fans to be more aggressive

caseyjlaw, 2023-09-28:
Nice work. By my eye, there is a 5C improvement with "full" fan mode.
The fact that there is a 15C difference between hottest and coolest server is interesting. Perhaps there is a bit we can do with airflow to shrink that range.

yupinghuang, 2023-12-01:
I cranked up the fan to full speed on all nodes. That worked.

closed 2023-12-01 by yupinghuang