ovro-lwa/lwa-issues #480: Account/system access for Craig Taylor

jaycedowell, 2023-09-28:
I have a student, @ctaylor-physics, who is wanting to run a fringe test between the NM LWA stations and OVRO.  He'll need an account and I'd also like for him to be able to access the pipeline account on calim and the ubuntu account on lwateng.

caseyjlaw, 2023-09-28:
I'll send the New User tutorial. Kemby can help get the account set up if you share your public key here.

ctaylor-physics, 2023-10-03:
Hello, here is my public key for one of our machines.
[cat_virgo_publickey.zip](https://github.com/ovro-lwa/lwa-issues/files/12795321/cat_virgo_publickey.zip)

kembyct, 2023-10-04:
Can you paste it here.
Thanks.

ctaylor-physics, 2023-10-04:
Sure thing, here it is:

ssh-rsa AAAAB3NzaC1yc2EAAAADAQABAAACAQDDMdJNgocd0hMomN4vKIFfy4egI8XHZD56+JfmzR/z3a2pHsoi0D6zP558fRb5JhoUo5PRrOzV9MP2kzkshbn/fRcGHFd7u+BzeAdYDblClA6b6cNu7eWX57/nXAQ94e5/UWo3/TWkqKeUStACIOYNUQAEX/5e9sreHRdMIe1QHwu0PUood3pRDLbmV2C6TZAg32RJ/YUfWtjyX3p5Giv3IasOb0KqMBt1wcJMpNSflmclndwUQqQ/2ow8ynzL+kTFXow21JpFQOzjN7vpikawBxOAXC6ljRf+/4KIute74KiU+51/oqnpoVu3mtVmkOIzwTgBbQ8OOz43wJLNSBWiN+RaR3i6LoAibDjuffQ0qtP3vRTbM1RVY0F4cWu+20JWiF4nGuc/iCPOUr6tlVxn05scP6nVpomvNYdb3098V7V5bggS/W+N21EAkQw5YyKggkLMdAMqc3/TqFFUMCCCGQlrFzE/GpZxkYuf/f3+WOpA2C1exzYZr1V5o354EvYzJOBkbMlE2djfeFk2prUfs4FRbj23NnU5UvdAgoX4lUuvXdcFWiNi+O317fXZGGyPlIuHv4XY1aWe7KdDcfkUtO6N23G8EoDltczFLD6nEQ8tNyywkyopzxqBE9I6EEEYxexS98X6NHzwQNhrs5b+YklQYBm05EdWJI27woFf6w== ctaylor@virgo

kembyct, 2023-10-04:
Great
Can I have your first and last name, and a preferred username ?

ctaylor-physics, 2023-10-04:
Craig Taylor, 'ctaylor' is the username that I have for all the machines here at UNM. so if that is available I'd use that. Otherwise "cataylor" or "ctaylor" with a number at the end is fine.

kembyct, 2023-10-04:
Done!
Let me know if you have any question.

ctaylor-physics, 2023-10-04:
Hey Kemby, I am trying to get into ubuntu@lwamaas to test the new logins and I get publickey denial. Can you help me get in?

-bash-4.2$ ssh ubuntu@lwamaas.lwa.pvt
Permission denied (publickey).

kembyct, 2023-10-04:
I think you should "ssh -YA ctaylor@ssh.ovro.caltech.edu", then ssh to ubuntu@lwamaas.lwa.pvt

ctaylor-physics, 2023-10-09:
This worked, thanks Kemby. I should have access to all the points that Jayce wants.

kembyct, 2023-10-09:
Sure.

closed 2023-10-09 by jaycedowell