ovro-lwa/lwa-issues #486: Fast visibility YY polarisation has no fringes

surajit4444mondal, 2023-10-11:
I have prepared some slides to backup my claim. All data and images are available at calim7:/data07/msurajit/fast_vis_YY_test/ .
[fast_vis_updates.pdf](https://github.com/ovro-lwa/lwa-issues/files/12871967/fast_vis_updates.pdf)

jack-h, 2023-10-11:
Has this always been an issue and it has just been noticed, or is this new?

surajit4444mondal, 2023-10-11:
The earliest data I have for fast visibilitites come from June 2023, and this issue is there as well. So my guess is this is not a recent issue and has been there all along. It is only now that we have found it.

caseyjlaw, 2023-10-12:
We'll use f-eng test vector to see how fast vis amplitude changes.

yupinghuang, 2023-10-12:
Actually, if the antenna selection is off, one should still see fringes in frequency.

jack-h, 2023-10-12:
Note that there are 2 types of built in test vectors, one which is varies with frequency but not by antpol, the other varies by antpol but not freq (per board). Or, you can provide your own vectors.

[image]

https://github.com/realtimeradio/caltech-lwa/blob/master/docs/build/latex/LWA352SNAP2F-Engine.pdf

caseyjlaw, 2023-10-12:
@surajit4444mondal In Python:
```
from lwa_f import snap2_fengine, snap2_feng_etcd_client
f = snap2_fengine.Snap2Fengine(name)  # "snap02"
f.input.get_status()  # print current input type for that snap
f.eqtvg.write_const_per_stream()
f.eqtvg.write_freq_ramp()
```

If `stream=None` then all inputs to that SNAP will be changed. And then is there a `f.eqtvg.tvg_enable()` and `tvg_disable()`, I think? Python should help you see the methods and some docs.

yupinghuang, 2023-10-12:
autocorrs XX vs YY
[image]

autocorrs XX vs YX
[image]

autocorr YX vs XY

might be worth trying YX in your calibrated fast vis and see if it looks more like YY (assuming that the X and Y solutions aren't too different). Or maybe edit the measurement set to swap the last two elements of the last dimensions of each row (i.e. YX and YY).

surajit4444mondal, 2023-10-12:
@yupinghuang  I actually tied to swap YY and YX, but I did not see the fringes. I do that test today again, just to ensure that I did not do anything wrong..  The auto-correlation plots you showed are clearly encouraging..

yupinghuang, 2023-10-12:
Sooooooooooooooooooooooooooooooooooooooo nope it's not that simple.

[image]

jack-h, 2023-12-15:
Sure -- as ever the docs at https://github.com/realtimeradio/caltech-lwa/blob/master/docs/build/latex/LWA352SNAP2F-Engine.pdf should help.

1. Start with a pre-initialized board (`cold_start` previously run)
2. Set the input to use on-board noise generators for all SNAP boards: `fengine.input.use_noise()`
3. [Maybe necessary] Retune the EQ levels, since the noise generator probably has quite different power levels to the real sky inputs: `fengine.set_equalization(start_chan=0, stop_chan=4095)`
4. You could just take data now, and you should see the same correlation between all antennas and polarizations. You could also to the below, to force a phase slope over the band...

Optional extras:
6. Set the delays of each antenna to have a slightly different time delay `fengine.delay.set_delay(<stream id>, <delay>)`, where `stream id` is the input number on this SNAP board and runs from 0 to 63. I think all the even inputs are one polarization, and all the odd IDs the other. One way to do this might be to use `delay = stream id` for each input of a board.
7. Take correlator data! You should see a phase slope between antennas, corresponding to the relative delays between them. One cycle of delay yields 180 degrees over the full Nyquist band.

As an additional comment, it wouldn't hurt to have different test vectors per polarization and per freq channel in your original test (though you mentioned that the channel assignments had already been verified via  a "ramp" test vector run.

yupinghuang, 2023-12-22:
@surajit4444mondal I heard that this got resolved? Mind letting us know what happened?

surajit4444mondal, 2023-12-22:
@yupinghuang This has not been resolved yet. What ever updates I had are already present in this issue.. We found that for a static signal things are working as expected. Jack suggested that it is possible that the timestamps are getting mixed somehow. So the plan is to do a noise-injection test. I will update this issue when we have results of that test.