ovro-lwa/lwa-issues #488: improve deployment of ovro_data_recorder

caseyjlaw, 2023-10-12:
ovro_data_recorder runs services by executing script/modules in the base of the repo. This makes deployment fragile, since pulling code will automatically get deployed. Unrelated: we had confusion related to running recorder services for all recorders on a node, when it should be limited to set of recorders per server.
It would be good to consider a new repo structure and installation/service scheme.

jaycedowell, 2023-11-09:
I've added `version` monitor point that is updated when a recorder starts.  It currently records the branch in use, the hash of the commit in use, and whether or not the repo has uncommitted changes.  So:
```
from mnc.mcs import Client
c=Client()
c.read_monitor_point('version', 'dr2')
```
yields:
```
<MonitorPoint timestamp='2023-11-07 16:22:44.034964',
    value='main@6c20b13', unit=''>
```.
That's not everything that is needed but it's a start.

With regard to the multiple recorders that seemed to be related to https://github.com/lwa-project/ovro_data_recorder/issues/38.

jaycedowell, 2023-12-18:
See https://github.com/lwa-project/ovro_data_recorder/pull/44

caseyjlaw, 2023-12-19:
Looks great! If you are happy with the way it works, feel free to close this.

jaycedowell, 2023-12-19:
It needs to be deployed on calim.  I think that's the last thing.

closed 2023-12-20 by jaycedowell