ovro-lwa/lwa-issues #489: fast vis still not working on lustre

yupinghuang, 2023-10-17:
[image]

at about 800% gulp time.

jaycedowell, 2023-10-17:
Is the plan for everything (slow, fast, power beams) to move to /lustre?

yupinghuang, 2023-10-17:
Yes.

It might be worth removing the flush calls and see what happens. We know for sure that there's more than enough throughput to go around, but if there are synchronized read/write calls, the latency will be tricky.

caseyjlaw, 2023-11-09:
@jaycedowell did your commit to the data recorder resolve this issue?
Also, how do you feel about having data recorders writing to lustre directly? Do you anticipate any issues?

jaycedowell, 2023-11-09:
I don't know that writing to lustre has been tried again.  I haven't had a chance to try it.

caseyjlaw, 2023-11-10:
@yupinghuang reported that there was an issue with python threading and the GIL not working well with high-latency writes to lustre. Could be a threading/multiprocessing change could get this to work.

jaycedowell, 2023-11-29:
This also seems to be working after the latest round of updates.

caseyjlaw, 2023-11-29:
Terrific!

closed 2023-11-29 by caseyjlaw