ovro-lwa/lwa-issues #491: I might have started the snap etcd service in a weird place but...

yupinghuang, 2023-11-01:
this is after `initialize`. Restarting the service and then cold_start_from_config ran fine.
```
Nov 01 00:04:57 lwacalim10 lwa_start_feng_[2063328]: 2023-11-01 00:04:57,746 - /opt/devel/pipeline/envs/deployment/bin/lwa_start_feng_etcd_service.py:snap01 - ERROR - Command failed
                                                     Traceback (most recent call last):
                                                       File "/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/lwa_f/snap2_feng_etcd_client.py", line 564, in _etcd_callback
                                                         resp = cmd_method(**cmd_kwargs)
                                                       File "/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/lwa_f/snap2_fengine.py", line 544, in cold_start_from_config
                                                         dests = dests,
                                                       File "/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/lwa_f/snap2_fengine.py", line 701, in cold_start
                                                         self.eth.configure_source(mac, source_ip, source_port)
                                                       File "/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/lwa_f/blocks/eth.py", line 179, in configure_source
                                                         raise RuntimeError("Couldn't get Ethernet CasperFpga object")
                                                     RuntimeError: Couldn't get Ethernet CasperFpga object
Nov 01 00:04:57 lwacalim10 lwa_start_feng_[2063328]: 2023-11-01 00:04:57,753 - /opt/devel/pipeline/envs/deployment/bin/lwa_start_feng_etcd_service.py:snap01 - ERROR - Responded to command 'cold_start_from_confi>
Nov 01 00:06:08 lwacalim10 lwa_start_feng_[2063328]: 2023-11-01 00:06:08,890 - lwa_f.blocks.block:snap01 - WARNING - pfb - Firmware implements some hardcoded shift stages. Setting shift to 0x1fff
Nov 01 00:06:08 lwacalim10 lwa_start_feng_[2063328]: 2023-11-01 00:06:08,947 - lwa_f.blocks.block:snap01 - WARNING - eth - Couldn't find Ethernet core. Will retry later
Nov 01 00:06:08 lwacalim10 lwa_start_feng_[2063328]: 2023-11-01 00:06:08,947 - /opt/devel/pipeline/envs/deployment/bin/lwa_start_feng_etcd_service.py:snap01 - ERROR - Command failed
                                                     Traceback (most recent call last):
                                                       File "/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/lwa_f/snap2_feng_etcd_client.py", line 564, in _etcd_callback
                                                         resp = cmd_method(**cmd_kwargs)
                                                       File "/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/lwa_f/snap2_fengine.py", line 544, in cold_start_from_config
                                                         dests = dests,
                                                       File "/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/lwa_f/snap2_fengine.py", line 701, in cold_start
                                                         self.eth.configure_source(mac, source_ip, source_port)
                                                       File "/opt/devel/pipeline/envs/deployment/lib/python3.6/site-packages/lwa_f/blocks/eth.py", line 179, in configure_source
                                                         raise RuntimeError("Couldn't get Ethernet CasperFpga object")
                                                     RuntimeError: Couldn't get Ethernet CasperFpga object
Nov 01 00:06:08 lwacalim10 lwa_start_feng_[2063328]: 2023-11-01 00:06:08,954 - /opt/devel/pipeline/envs/deployment/bin/lwa_start_feng_etcd_service.py:snap01 - ERROR - Responded to command 'cold_start_from_confi>
```

yupinghuang, 2023-11-01:
if `cold_start_from_config` fails it returns an empty dict
```
In [11]: resp = ec.send_command(1, 'feng', 'cold_start_from_config', kwargs={'config_file': '/home/pipeline/proj/lwa-shell/mnc_python/config/lwa_config_calim.yaml', 'program': False, 'initialize': True}, timeout
    ...: =120)
Command status from FID 1 returned: 'error'

In [12]: resp
Out[12]: {}
```

and if it succeeds...it returns...one `None`?
```
In [14]: resp = ec.send_command(1, 'feng', 'cold_start_from_config', kwargs={'config_file': '/home/pipeline/proj/lwa-shell/mnc_python/config/lwa_config_calim.yaml', 'program': False, 'initialize': True}, timeout
    ...: =120)

In [17]: resp is None
Out[17]: True
```

I guess it make sense on some cosmic level but it can be hard to handle errors from these command.

jack-h, 2023-12-01:
Yeah, fair. If I make it return `OK` or `FAIL` will that break things elsewhere?

yupinghuang, 2023-12-01:
"Yeah, fair. If I make it return OK or FAIL will that break things elsewhere?"

Depends on what "it" is? If you're making the power monitoring and autocorr apis return that, they might break existing things. But if it's just the cold_start_from_config and program calls that we're talking about, it won't break things that are already working.

Any chance you can make the single snap board response consistent with when a command is sent to multiple boards?
Like, if I send_command to `0`, I expect a dict with a certain number of keys back. And if a snap number is not one of the keys, I know that it fails there.

So if I send a command to `3`, can I get a `{3: None}` back if it succeeds and an empty dict back if it fails?

jack-h, 2023-12-02:
I'll do two things:
1. Change `cold_start` and `cold_start_from_config` to return values.
2. Change the `ec.send_command` to return consistently between single and multiple boards, as requested. I expect that this will break some things in RFI scripts, but I can fix that as/when it happens.

Thanks for the series of ops. I'll try and recreate. Certainly `program` potentially takes so long to complete that it might freak out the etcd client if it is having to queue up lots of other requests from other people. Also, possibly the power cycling of the boards (which reverts firmware to a vanilla boot image) might just freak the etcd client out to an extent it can't recover.