ovro-lwa/lwa-issues #492: Improve reliability of data recording

dgary50, 2023-11-11:


dgary50, 2023-11-11:
This morning all recorders were reset and were working fine initially, but after less than 1 hour we get this.  When this occurs, the recorders cannot recover and it requires a reset of recorders on that node.

[image]

yupinghuang, 2023-11-16:
@jaycedowell 

so i'd suggest ditching quota for now. I'm fairly confident that for recording to the local disks, things should work about 80% of the time after I up the processes priority.

For lustre -- I think I can get a way to get disk use more quickly soon. Also there's more space to fill up.

dgary50, 2023-11-16:
Yuping, it looks like a bad idea to ditch the quota.  We are having a nice
solar event right now, but the calim1 disk is full and the beamforming
recorder stopped, so we have not spectrogram.  What was your expectation
about disks getting full?  Or maybe the quota is supposed to be running and
it somehow failed?

Dale

dgary50, 2023-11-16:
Or maybe Jayce is using calim1 for testing as he suggested earlier.  I now
realize we should have changed to a different beam in that case.

Dale

yupinghuang, 2023-11-16:
> jaycedowell__;!!DLa72PTfQgg!NY7xkipPBBcR4ju2Out4TqbEvhM5mh
> lwa-issues/issues/492*issuecomment-1815224755__;Iw!!DLa72PTfQgg!
> NY7xkipPBBcR4ju2Out4TqbEvhM5mhczfyZdmDfpZ5h2d3Q33P76NQlQzyZS
> VCNFSM6AAAAAA7HLEYF2VHI2DSMVQWIX3LMV43OSLTON2WKQ3PNVWWK3TUHM
> YTQMJVGIZDINZVGU__;!!DLa72PTfQgg!NY7xkipPBBcR4ju2Out4TqbEvhM5mh

Quotas are running. But I think jayce’s testing on calim01 may have
interfered with quota on the local disk.

jaycedowell, 2023-11-17:
Yeah, that is from my testing.  I had forgot that the current quota management is shared between the vis. recorders on that node.  By testing with dr-vslow-1 and dr-vfast-1 the quota enforcement was stopped for dr-vslow-2 and dr-vfast-2 and things filled up.

jack-h, 2023-11-17:
This seems pleasantly unrelated to the TX side, so I'm deassigning. Feel free to add me back in if it turns out this isn't just related to RX-side / quotas

jaycedowell, 2023-12-13:
The latest problem on calim03 seems to be from a heavy load on that node.  The one minute load average hit ~220 according to the power beam logs.