ovro-lwa/lwa-issues #493: [Account] for Peijin Zhang

peijin94, 2023-11-22:
Join as a new staff in NJIT solar group working on OVRO-LWA solar data processing

My preferred user name is "peijin", if not available then "peijin94"

My id_rsa.pub is
```
ssh-rsa AAAAB3NzaC1yc2EAAAADAQABAAABAQDggL39mIt5+3Y9Q2YT7Ya6eAvAreNfKD50sMRVoRNTJipJ5zUxtNVq7YKwvt1AfPaXAUMp8k41sfmH3kJ/lTOANlbcM0ng4PjlG7pXU9skWTynFJ+6mRWrqfgrXMYzHAd9D5D0R/2LUmlMARVymruqtpL3CC6YRoBUWNYUmX2jG1Ssoflndw3ThmThQ7GT5rsaBzSsEWhYLvLIcZbHYpCFiO9XBomDOjwOO1cpcIrugc8sDInQKw/jXlB78PJUiB1yEqiM/En0knInM1WhGTH+rKNTADwnNDdqIocT4dxOYYalaJ3y0HRKRPmytIHCMAnRlG3p7H3Duip9WD/tK2Yr peijin@ustc.edu
```

kembyct, 2023-11-29:
Hello Casey

Our internet has been down since yesterday. I'll take care of it as soon as we get it back.

thanks,
Kemby

Assigned #493<https://github.com/ovro-lwa/lwa-issues/issues/493> to @kembyct<https://github.com/kembyct>.

peijin94, 2023-11-30:
Follow up on this thread, still waiting for access to calim

yupinghuang, 2023-12-04:
I added Peijin's key to lwamaas and made an account for him on calim @kembyct so all he needs now is a ssh.ovro.caltech.edu account.

peijin94, 2023-12-06:
Name is Peijin Zhang, username prefer to be peijinz

kembyct, 2023-12-06:
It's done
You should be able to connect anytime. Let me know if you have a question.

Thanksj

caseyjlaw, 2023-12-14:
I believe the account is all set now.

closed 2023-12-14 by caseyjlaw