ovro-lwa/lwa-issues #499: dashboard ARX tab channel numbers off by 1

yupinghuang, 2023-12-04:
I don't know where the docs for the arx api is but the configuration spreadsheet and the rfimigitation plots suggest that arx channels start at 1.

rh-codebase, 2023-12-04:
All library type API's better be 0-based.... unless maybe fortran ones. If 1-based numbering is what a human want to relate to then that conversion should be done in the presentation layer. What dashboard is being referred to here?

yupinghuang, 2023-12-04:
https://github.com/ovro-lwa/mnc_python/blob/main/mnc/myarx.py#L149

the dashboard I'm referring to is Casey's jupyter-notebook-on-panel running as a web service on lwacalim10:5006

caseyjlaw, 2023-12-04:
Yep. That choice is already made, unfortunately.
I can make sure the health dashboard is consistent with naming used. Deploying shortly.

closed 2023-12-04 by caseyjlaw