ovro-lwa/lwa-issues #501: Fast visibility scaling

surajit4444mondal, 2023-12-15:
It seems that there is a factor of 4 difference between the fast and slow visibility. I have confirmed this using a test-vector test as well. While the predicted visibilities match exactly with the observed ones in case of slow visibilities, the observed fast visibilities were a factor of 4 higher than the predicted ones. The easiest way to explain this factor of 4 is the fact that the number of channels are different between fast and slow vis by exactly factor of 4. The observed fast and slow vis plots will clearly show that the range of slow visibilities is smaller than that for the fast visibilities. Now since this is a constant factor, this can be corrected by combining this in the bandpass calibration step. @binchensun @dgary50 What do you think? Should we bring this up or shall we correct for this in the bandpass correction for fast vis?
[image]
[image]

dgary50, 2023-12-15:
Good point. I am sure you are right about where the factor comes from.  I would mention it since others may want to use such data. I suppose it could be removed prior to recording, since it will be common practice to apply calibration from slow vis—it is better if there are no “magic” factors to apply. Regards,DaleSent from my iPhone

fast_vis_full.png (view on web)
slow_vis_full.png (view on web)

binchensun, 2023-12-15:
I agree.

closed 2023-12-19 by surajit4444mondal