ovro-lwa/lwa-issues #503: faulty drive on lwastor04

yupinghuang, 2023-12-17:
```
  pool: lwastor04
 state: DEGRADED
status: One or more devices are faulted in response to persistent errors.
	Sufficient replicas exist for the pool to continue functioning in a
	degraded state.
action: Replace the faulted device, or use 'zpool clear' to mark the device
	repaired.
config:

	NAME                        STATE     READ WRITE CKSUM
	lwastor04                   DEGRADED     0     0     0
	  raidz2-0                  DEGRADED     0     0     0
	    wwn-0x5000cca2b4cbbf58  ONLINE       0     0     0
	    wwn-0x5000cca2ab47c278  ONLINE       0     0     0
	    wwn-0x5000cca2b6601a98  ONLINE       0     0     0
	    wwn-0x5000cca2b4c9c29c  ONLINE       0     0     0
	    wwn-0x5000cca2b665fe8c  ONLINE       0     0     0
	    wwn-0x5000cca2b4ca4598  ONLINE       0     0     0
	    wwn-0x5000cca2b662085c  ONLINE       0     0     0
	    wwn-0x5000cca2b4c7d170  ONLINE       0     0     0
	    wwn-0x5000cca2b4c9fdf4  ONLINE       0     0     0
	    wwn-0x5000cca2ab4c24ec  FAULTED     15     0     0  too many errors
	    wwn-0x5000cca2b65bb7d8  ONLINE       0     0     0
	    wwn-0x5000cca2b6659e30  ONLINE       0     0     0
```

yupinghuang, 2023-12-17:
kernel log. zfs is being truly excellent and lustre is still working (and not writing data to that vdev).

the fact that we only got read errors made me think that it might have been a transient communication thing. A `smartctl` short test came back alright on the drive. A long test is running right now.

```
[Sun Dec 17 10:10:57 2023] sd 0:0:9:0: attempting task abort!scmd(0x000000009d2b5806), outstanding for 30480 ms & timeout 30000 ms
[Sun Dec 17 10:10:57 2023] sd 0:0:9:0: [sdl] tag#4404 CDB: Read(16) 88 00 00 00 00 00 5d d7 62 da 00 00 00 1a 00 00
[Sun Dec 17 10:10:57 2023] scsi target0:0:9: handle(0x0030), sas_address(0x5000cca2ab4c24ed), phy(21)
[Sun Dec 17 10:10:57 2023] scsi target0:0:9: enclosure logical id(0x50015b2140b646bf), slot(9) 
[Sun Dec 17 10:10:57 2023] scsi target0:0:9: enclosure level(0x0000), connector name( C0.1)
[Sun Dec 17 10:10:57 2023] sd 0:0:9:0: task abort: SUCCESS scmd(0x000000009d2b5806)
[Sun Dec 17 10:10:57 2023] sd 0:0:9:0: [sdl] tag#4404 FAILED Result: hostbyte=DID_TIME_OUT driverbyte=DRIVER_OK cmd_age=30s
[Sun Dec 17 10:10:57 2023] sd 0:0:9:0: [sdl] tag#4404 CDB: Read(16) 88 00 00 00 00 00 5d d7 62 da 00 00 00 1a 00 00
[Sun Dec 17 10:10:57 2023] blk_update_request: I/O error, dev sdl, sector 12595173072 op 0x0:(READ) flags 0x700 phys_seg 2 prio class 0
[Sun Dec 17 10:10:57 2023] zio pool=lwastor04 vdev=/dev/disk/by-id/wwn-0x5000cca2ab4c24ec-part1 error=5 type=1 offset=6448720224256 size=106496 flags=180980
```

kembyct, 2023-12-18:
Hello Yuping

It's cleared and the disk is getting repaired.

Thanks,
Kemby

yupinghuang, 2023-12-18:
Hi Kemby,

How could you tell that it was cleared? I saw someone running `zpool clear` and I assume that's why it went back online. 

The smartctl long test has not finished.

kembyct, 2023-12-18:
Yes
I run "zpool clear lwastor04", the particular went to resilvering mode, which fixed all the errors. Now it display no error when you run  "zpool satus -x"

Thanks

rh-codebase, 2023-12-18:
I'm not sure 'zpool clear ...' actually solves the problem or just 'clears' the errors. @kembyct can you provide more information as to what zpool clear actually is doing?
I was under the impression that ZFS self heals and does not need a resilver to fix corrupted data. How often should scrubs be done? What's the performance hit on scrubs?  Somewhat related, are you keeping track of which disks are having issues so we can determine the efficacy of solutions?

@Yuping. Trust in smartctl? How much faith should be placed in this tool? I remember Google did an analysis related to whether there were any metrics which could reliably be used to predict disk failure and they didn't find any. They included smartctl results. I'll see if I can dig this up since I may be misinterpreting the takeaways from the report.

Maybe this is all that can be done until a disk ultimately fails as in 'no response'. I think it's prudent for Kemby to set up some type of real-time monitoring system to allow performance related issues to be resolved quickly. Other thoughts?

rh-codebase, 2023-12-18:
@kemby. In the logs Yuping showed,:

|[Sun Dec 17 10:10:57 2023] zio pool=lwastor04 vdev=/dev/disk/by-id/wwn-0x5000cca2ab4c24ec-part1 error=5 type=1 offset=6448720224256 size=106496 flags=180980

can you find out what it actually means.  For ex: error=5, type=1 flags=180980

yupinghuang, 2023-12-18:
I think having a few disks stashed somewhere as spares would be a great idea. And right now if I have a degraded OST alert to slack (which usually comes from zfs marking a pool unhealthy like in this case) over on grafana.

The safest thing to do in a vacuum is to replace the disk (and also reconnect the cable in the meantime). But then person needs to access the rack etc etc.
 
I ran smartctl to see if there was a sector failure or something that's obviously wrong with the disk (or the cable) that screams replacement.

I do **not** believe that we should run zpool clear before we are confident that the disk is okay. Resilvering assumes that you have an okay drive in there -- it probably checksums on records to determine which records need to be copied. The disk driver may decide to remap bad blocks it know about. If the OST slows down dramatically in the coming days, we probably need to replace the disk.

I have very little experience with harddrives so I dunno what to do.

rh-codebase, 2023-12-18:
I've never seen any real convincing arguments as to what to do with harddrives. My experience seems to come down to what one can afford to do. If LWA has the money to replace drives on every hint of failure then do it.  

I agree with you about resilver. My understanding is that 'scrub' should be looked into. The tool reports: 'scan: scrub repaired 0 in 0h2m with 0 errors on Tue Feb 15 09:04:04 2022'
seems to be what is needed and hints that it is capable of repairing corrupted data.

I think Mark finally did get spares ordered. @kemby: Do you have spare disks? How many? These are all disks < 5yrs old so should  be under warranty. Please look into what is needed to do on our side to return a disk under warranty. Some manuf. have scanning tools so maybe the procedure is to replace the disk outright and use manuf. tools to determine if the disk is returnable. What's your take on this?

yupinghuang, 2023-12-18:
smartctl long test came back negative.

I am gonna run a `https://openzfs.github.io/openzfs-docs/man/master/8/zpool-scrub.8.html` on `lwastor04` (i.e. the problem pool) and see how long that takes.

"Scrubbing and resilvering are very similar operations. The difference is that resilvering only examines data that ZFS knows to be out of date (for example, when attaching a new device to a mirror or replacing an existing device), whereas scrubbing examines all data to discover silent errors due to hardware faults or disk failure."

rh-codebase, 2023-12-18:
Do you have a way to determine performance hit if any during the scrub? That's one nice thing IMO about ZFS, scrub only looks at data, not the entire disk so scrubs can be fast on large disks with little data. I would vote for as many scrubs as can be tolerated

@kemby. Can you look into tools which can collected the scrub information... or write one?

kembyct, 2023-12-18:
When I run zpool clear, It displayed, (resilvering) on the the faulted disk for a few second before it start displaying  all pool are healthy running "zpool status -x". I think degraded disk is in the better shape than the failed one, that's what ZFS fixed it with "zpool clear". To get more details on the errors, we'll have to run scrub a couple time to see how many read and checksum errors on the disk. I have not run scrub yet. I'm about to do it right away

yupinghuang, 2023-12-18:
I’ll leave the scrub to y’all then. I/O out of each pool is on grafana
under “Lustre Overview”, loads across all OSTs is typically quite blanced

kembyct, 2023-12-18:
zpool status -v lwastor04
  pool: lwastor04
 state: ONLINE
  scan: resilvered 574M in 00:01:04 with 0 errors on Mon Dec 18 17:21:34 2023
config:

        NAME                        STATE     READ WRITE CKSUM
        lwastor04                   ONLINE       0     0     0
          raidz2-0                  ONLINE       0     0     0
            wwn-0x5000cca2b4cbbf58  ONLINE       0     0     0
            wwn-0x5000cca2ab47c278  ONLINE       0     0     0
            wwn-0x5000cca2b6601a98  ONLINE       0     0     0
            wwn-0x5000cca2b4c9c29c  ONLINE       0     0     0
            wwn-0x5000cca2b665fe8c  ONLINE       0     0     0
            wwn-0x5000cca2b4ca4598  ONLINE       0     0     0
            wwn-0x5000cca2b662085c  ONLINE       0     0     0
            wwn-0x5000cca2b4c7d170  ONLINE       0     0     0
            wwn-0x5000cca2b4c9fdf4  ONLINE       0     0     0
            wwn-0x5000cca2ab4c24ec  ONLINE       0     0     0
            wwn-0x5000cca2b65bb7d8  ONLINE       0     0     0
            wwn-0x5000cca2b6659e30  ONLINE       0     0     0

errors: No known data errors

rh-codebase, 2023-12-18:
Kemby, keep an eye on the performance metrics. They may dictate how often scrubs can be run.

kembyct, 2023-12-18:
Sure. The scrub has started.
sudo zpool scrub lwastor04
[ubuntu@lwastor04 ~]$ zpool status -v lwastor04
  pool: lwastor04
 state: ONLINE
  scan: scrub in progress since Mon Dec 18 20:23:45 2023
        3.20G scanned at 273M/s, 2.88M issued at 246K/s, 50.0T total
        0B repaired, 0.00% done, no estimated completion time
config:

        NAME                        STATE     READ WRITE CKSUM
        lwastor04                   ONLINE       0     0     0
          raidz2-0                  ONLINE       0     0     0
            wwn-0x5000cca2b4cbbf58  ONLINE       0     0     0
            wwn-0x5000cca2ab47c278  ONLINE       0     0     0
            wwn-0x5000cca2b6601a98  ONLINE       0     0     0
            wwn-0x5000cca2b4c9c29c  ONLINE       0     0     0
            wwn-0x5000cca2b665fe8c  ONLINE       0     0     0
            wwn-0x5000cca2b4ca4598  ONLINE       0     0     0
            wwn-0x5000cca2b662085c  ONLINE       0     0     0
            wwn-0x5000cca2b4c7d170  ONLINE       0     0     0
            wwn-0x5000cca2b4c9fdf4  ONLINE       0     0     0
            wwn-0x5000cca2ab4c24ec  ONLINE       0     0     0
            wwn-0x5000cca2b65bb7d8  ONLINE       0     0     0
            wwn-0x5000cca2b6659e30  ONLINE       0     0     0

errors: No known data errors

kembyct, 2023-12-18:
Update:
The scrub will be completed in 6 hours.

scrub in progress since Mon Dec 18 20:23:45 2023
        50.0T scanned at 0B/s, 15.5T issued at 1.48G/s, 50.0T total
        0B repaired, 30.94% done, 06:38:46 to go
config:

        NAME                        STATE     READ WRITE CKSUM
        lwastor04                   ONLINE       0     0     0
          raidz2-0                  ONLINE       0     0     0
            wwn-0x5000cca2b4cbbf58  ONLINE       0     0     0
            wwn-0x5000cca2ab47c278  ONLINE       0     0     0
            wwn-0x5000cca2b6601a98  ONLINE       0     0     0
            wwn-0x5000cca2b4c9c29c  ONLINE       0     0     0
            wwn-0x5000cca2b665fe8c  ONLINE       0     0     0
            wwn-0x5000cca2b4ca4598  ONLINE       0     0     0
            wwn-0x5000cca2b662085c  ONLINE       0     0     0
            wwn-0x5000cca2b4c7d170  ONLINE       0     0     0
            wwn-0x5000cca2b4c9fdf4  ONLINE       0     0     0
            wwn-0x5000cca2ab4c24ec  ONLINE       0     0     0
            wwn-0x5000cca2b65bb7d8  ONLINE       0     0     0
            wwn-0x5000cca2b6659e30  ONLINE       0     0     0

kembyct, 2023-12-19:
Update:
 zpool status -v lwastor04
  pool: lwastor04
 state: ONLINE
  scan: scrub in progress since Mon Dec 18 20:23:45 2023
        50.0T scanned at 0B/s, 43.4T issued at 1.35G/s, 50.0T total
        0B repaired, 86.92% done, 01:22:33 to go
config:

        NAME                        STATE     READ WRITE CKSUM
        lwastor04                   ONLINE       0     0     0
          raidz2-0                  ONLINE       0     0     0
            wwn-0x5000cca2b4cbbf58  ONLINE       0     0     0
            wwn-0x5000cca2ab47c278  ONLINE       0     0     0
            wwn-0x5000cca2b6601a98  ONLINE       0     0     0
            wwn-0x5000cca2b4c9c29c  ONLINE       0     0     0
            wwn-0x5000cca2b665fe8c  ONLINE       0     0     0
            wwn-0x5000cca2b4ca4598  ONLINE       0     0     0
            wwn-0x5000cca2b662085c  ONLINE       0     0     0
            wwn-0x5000cca2b4c7d170  ONLINE       0     0     0
            wwn-0x5000cca2b4c9fdf4  ONLINE       0     0     0
            wwn-0x5000cca2ab4c24ec  ONLINE       0     0     0
            wwn-0x5000cca2b65bb7d8  ONLINE       0     0     0
            wwn-0x5000cca2b6659e30  ONLINE       0     0     0

kembyct, 2023-12-19:
Update:
zpool status -v lwastor04
  pool: lwastor04
 state: ONLINE
  scan: scrub repaired 0B in 10:54:43 with 0 errors on Tue Dec 19 07:18:28 2023
config:

        NAME                        STATE     READ WRITE CKSUM
        lwastor04                   ONLINE       0     0     0
          raidz2-0                  ONLINE       0     0     0
            wwn-0x5000cca2b4cbbf58  ONLINE       0     0     0
            wwn-0x5000cca2ab47c278  ONLINE       0     0     0
            wwn-0x5000cca2b6601a98  ONLINE       0     0     0
            wwn-0x5000cca2b4c9c29c  ONLINE       0     0     0
            wwn-0x5000cca2b665fe8c  ONLINE       0     0     0
            wwn-0x5000cca2b4ca4598  ONLINE       0     0     0
            wwn-0x5000cca2b662085c  ONLINE       0     0     0
            wwn-0x5000cca2b4c7d170  ONLINE       0     0     0
            wwn-0x5000cca2b4c9fdf4  ONLINE       0     0     0
            wwn-0x5000cca2ab4c24ec  ONLINE       0     0     0
            wwn-0x5000cca2b65bb7d8  ONLINE       0     0     0
            wwn-0x5000cca2b6659e30  ONLINE       0     0     0

errors: No known data errors

rh-codebase, 2023-12-19:
A post I found on Reddit. Might be worth remembering next time:

"

Whether a read error is acceptable or not really depends on its cause.

What you should do is first do a smart long test (smartctl -t long /dev/devicename) on the drive that's generating those errors. If the long test fails, then it will report the LBA where it failed, giving you a start point to do some testing (smartctl -a /dev/devicename).

What *I* usually do is check a few sectors before and after the reported LBA with hdparm.

Let's say the SMART self test states it failed reading at LBA 1999. I'll check sectors 1990 through 2010:

for i in {1990..2010};do hdparm --read-sector $i /dev/devicename;sleep 2s;done

This will test those 20 sectors to see if they can be read. If you get any kind of sense errors or inability to read those sectors, don't worry, hdparm will complain (note 2s sleep to give you time to see if an error gets thrown). However, if you get a nice blob of non-zero hexadecimal with no sense errors, then those sectors are probably fine.

If you get a bunch of sectors that are unreadable, then you should:

Run smartctl -a again, and note the current number of reallocated sectors. For the sake of argument, let's say you started with a raw value of "4".

Pick one of those unreadable sectors from before. Let's say 1993 was unreadable. You can do the following to write 0's to it:

hdparm --write-sector 1993 --yes-i-know-what-i-am-doing /dev/devicename

Before you (or anyone else) freaks out over this, remember that you aren't able to read that sector anyway, so there's no sense in giving a crap if you write a bunch of 0's to it.

The sector should get overwritten in a moment or longer. I've had some drives take over a minute for it to finally write over that "bad" sector. If it can't successfully overwrite that sector, it's time to start the replacement process.

If it CAN overwrite that sector, run another 'smartctl -a' against the drive and see if the reallocated sector raw value increased by one (in this example, goes from 4 to 5). If it has, then you likely have a small swath of sectors that need to get overwritten so they wind up re-allocated by reserve sectors. If your reallocated sector count did NOT increase, then the sector(s) in question are simply "fuzzy", still not *GREAT*, but you haven't burned a reserve sector. The sector is still good enough to hold data once it's overwritten (which you just did lol).

Note that if the sector was successfully overwritten, the next error that will be reported by zfs (not smartctl) against that sector will be a checksum error. It will no longer be a read error, but the checksum will be wrong because of the 0's you just overwrote into it. It'll get caught on the next scrub or read of that particular sector, so it will "heal" on its own, assuming we're talking about a raidz* or mirror vdev.

So, the big question is: If a drive is currently reallocating these bad sectors, is the drive still "good?" Maybe. Only you can decide whether the risk is acceptable. Fuzzy sectors or reallocated sectors are usually the result of drive age. If the drive is old, then this behavior is really to be expected. But there's no telling how long the drive might last. I have several drives in my 45 disk pool that had some fuzzy sectors. A couple of these drives are WD "Green" drives with well over 400 reallocated sectors over a couple of hundred events. Those drives are still in service today, have over 70,000 power on hours on them, and these problems started at the 45,000 hour mark, give or take, with no additional trouble since then. Other drives with far fewer hours on them started a rapid decline as soon as these errors started popping up. YMMV.

Personally, I don't care about the data on my particular pool. It's just video files I can always get back. So I find the risk to that data within acceptable limits for me. Your situation might be very different.

Edit: Also, to someone else's point about cabling, you should absolutely rule that out. Typically a cabling error will show up in 'dmesg' as an I/O error with *NO* preceding "critical medium error". If you see a critical medium error followed by a bunch of I/O errors, then that's definitely a bad/weak sector. If it's an I/O error *without* a critical medium error of some sort, then you might just have a cabling issue. Cabling issues are really common with home NAS setups. Even the best breakout cable is junk compared to the lowest-end backplane.
"

yupinghuang, 2023-12-19:
sweet!

I think I'm gonna have zfs scrub runs on monthly cadence (sequentially going through all the OSTs on each node) and make sure that they all run at the same time. I saw very little performance hit with the scrub that was just run

yupinghuang, 2023-12-19:
Still though: Do you have spare disks? How many? @kembyct

yupinghuang, 2023-12-19:
> sequential is a bit tricky since scrub is non-blocking and the time to finish depends on data quantity and loading, not disk size.

There's https://openzfs.github.io/openzfs-docs/man/master/8/zpool-scrub.8.html#w. And yeah we can keep an eye on that. Each pool has approximately the same amount of data. I'll see what `zfs-scrub.timer` for systemd says.

kembyct, 2023-12-19:
@yupinghuang I got about 4 spare disks

yupinghuang, 2023-12-22:
welp `lwastor01a` today had too many failures. Started `sudo smartctl -t long /dev/disk/by-id/wwn-0x5000cca2b6641874-part1`.

```
Fri Dec 22 19:25:16 2023] sd 0:0:17:0: attempting task abort!scmd(0x00000000b3a6481b), outstanding for 30431 ms & timeout 30000 ms
[Fri Dec 22 19:25:16 2023] sd 0:0:17:0: [sdt] tag#4337 CDB: Read(16) 88 00 00 00 00 00 a1 c7 e6 3a 00 00 00 1a 00 00
[Fri Dec 22 19:25:16 2023] scsi target0:0:17: handle(0x0038), sas_address(0x5000cca2b6641875), phy(29)
[Fri Dec 22 19:25:16 2023] scsi target0:0:17: enclosure logical id(0x50015b2140b7483f), slot(17) 
[Fri Dec 22 19:25:16 2023] scsi target0:0:17: enclosure level(0x0000), connector name( C0.1)
[Fri Dec 22 19:25:16 2023] sd 0:0:17:0: task abort: SUCCESS scmd(0x00000000b3a6481b)
[Fri Dec 22 19:25:16 2023] sd 0:0:17:0: [sdt] tag#4337 FAILED Result: hostbyte=DID_TIME_OUT driverbyte=DRIVER_OK cmd_age=30s
[Fri Dec 22 19:25:16 2023] sd 0:0:17:0: [sdt] tag#4337 CDB: Read(16) 88 00 00 00 00 00 a1 c7 e6 3a 00 00 00 1a 00 00
[Fri Dec 22 19:25:16 2023] blk_update_request: I/O error, dev sdt, sector 21713859024 op 0x0:(READ) flags 0x700 phys_seg 2 prio class 0
[Fri Dec 22 19:25:16 2023] zio pool=lwastor01a vdev=/dev/disk/by-id/wwn-0x5000cca2b6641874-part1 error=5 type=1 offset=11117487431680 size=106496 flags=180980
```

yupinghuang, 2023-12-25:
`smartctl` long test finished on Dec 23 second half of the day. I ran `zpool clear` on Dec 24 03:52.

On Dec 25, 13:12ish, same errors came back. I think we should get the cable checked for `wwn-0x5000cca2b6641874` on lwastor01.

yupinghuang, 2023-12-29:
Here's the list of events so far (with dmesg) https://github.com/ovro-lwa/calim-ops/wiki/Ongoing-list-of-OST-transient-outages. 6 incidents so far (and 2 on the same pool). They don't look like bad blocks since sometimes we get errors on multiple drives. And I think we need more incidents to figure out what's going on.

I've been bringing them back online since smartctl & scrub have consistently turned out fine. So.....theories

- something environmental? CPU temperature looked alright.
- some sort of weird power saving things?
- cabling or backplane or power supply? It'd be interesting to see if they only come up on certain chasis.
- it could still be a faulty batch of disks, which would be terrifying

yupinghuang, 2024-01-04:
we're up to 11 occurrences.

```
 lwastor06a
2024-01-01T09:06:33
lwastor01a
2023-12-22T19:25:16
2023-12-25T03:12:40
lwastor01b
2024-01-02T01:39:31
lwastor04
2023-12-17T10:10:57
2024-01-02T02:51:10
lwastor04a
2023-12-26T19:23:19
2024-01-04T08:27:28
lwastor07a
2023-12-28T22:27:18
2024-01-04T09:31:06
lwastor08
2023-12-24T10:07:53 
```

kembyct, 2024-01-05:
zpool iostat
              capacity     operations     bandwidth
pool        alloc   free   read  write   read  write
----------  -----  -----  -----  -----  -----  -----
lwastor01   83.1T   113T    266    337  22.2M  63.4M
lwastor01a  70.2T   126T    404    246  56.4M  47.9M
lwastor01b  91.1T   105T    240    308  19.7M  57.5M
----------  -----  -----  -----  -----  -----  -----
[ubuntu@lwastor01 ~]$ sudo zpool status -x
all pools are healthy

[ubuntu@lwastor02 ~]$ sudo zpool iostat
              capacity     operations     bandwidth
pool        alloc   free   read  write   read  write
----------  -----  -----  -----  -----  -----  -----
lwastor02   84.9T   112T    155    327  11.2M  48.7M
lwastor02a  79.0T   117T    168    316  11.7M  51.9M
lwastor02b  79.5T   117T    181    319  13.1M  52.9M
----------  -----  -----  -----  -----  -----  -----

All seem good.

yupinghuang, 2024-01-05:
Well that’s because I’ve been zpool clearing. I/O errors do come up. See
above for details

kembyct, 2024-01-05:
I wonder if we can track the specific jobs or processes that result in read error from the Lustre clients. Maybe operations between the network and CPUs need to be troubleshooted. I've only seen read errors so far, and ZFS is able to handle that. Checksum error affect data integrity, and disk replacement is more likely to be required in this event.

rh-codebase, 2024-01-10:
At least 2 hypotheses have been floated: bad cables, bad firmware. A table of which host slots have errors may show if there's a cable issue or issues. This can more easily be checked by swapping the drive in a 'bad' slot with a drive from another slot which has no errors. If the problem moves with the drive, then it's likely the drive if the slot error goes away. Does this seem reasonable?  For firmware, @kembyct , see if you can extract the firmware id on each drive. It should be included in the table which is counting the number of seen I/O related errors for each drive as a function of disk id, slot, host, time of day, disk temperature and anything else you can think of. If all the drives have the same firmware, then I would expect an even distribution of errors. It doesn't look like we are seeing this but maybe not enough stats. Any other thoughts on how best to go about testing these two hypotheses? Scanning dmesg on each host for past I/O error may help you populate this table. I've seen: zio, LNetError and Lustre error type messages in dmesg. save the current dmesg into a file and then after collecting the needed information you can clear it(dmesg -c). This can be done daily until we feel we have enough stats to see which path forward to take. 

Kemby, can you also look into whether you can get the controller card's temperature?

kembyct, 2024-01-12:
Update:
Firmware:
 version: C680
 product: WUH721818AL5204
 vendor: WDC
It's all the same across every node running these cmd(s) "lshw -c disk;  smartctl -a /dev/disk/by-id/wwn-0x500..."
The dmeg info was saved to dmesg1 file on each node and the buffer was cleared also. the drives temperatures are 24 for every disk. I'm still digging on the controller temperature and will update it as soon as possible.

kembyct, 2024-01-12:
for i in {1..8}; do ssh -YA lwastor0$i sudo zpool iostat -l; done
              capacity     operations     bandwidth    total_wait     disk_wait    syncq_wait    asyncq_wait  scrub   trim
pool        alloc   free   read  write   read  write   read  write   read  write   read  write   read  write   wait   wait
----------  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----
lwastor01   91.1T   105T    417    338  57.4M  61.8M   26ms   10ms    8ms    2ms  497us  487ns    2ms    8ms   41ms      -
lwastor01a  79.2T   117T    509    278  79.1M  52.8M   26ms   10ms    6ms    2ms   35us  473ns  736us    8ms   33ms      -
lwastor01b  98.6T  97.8T    736    311  55.6M  56.2M   16ms   10ms    5ms    2ms  409us  500ns    2ms    8ms   16ms      -
----------  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----
              capacity     operations     bandwidth    total_wait     disk_wait    syncq_wait    asyncq_wait  scrub   trim
pool        alloc   free   read  write   read  write   read  write   read  write   read  write   read  write   wait   wait
----------  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----
lwastor02   91.8T   105T    651    331  67.2M  51.2M   43ms    8ms    4ms    1ms  485us    1us    3ms    7ms   53ms      -
lwastor02a  87.4T   109T    351    325  65.5M  54.8M   50ms    9ms    7ms    1ms   72us    1us  832us    7ms   84ms      -
lwastor02b  87.9T   108T    223    329  24.8M  55.6M   32ms    9ms    9ms    2ms   66us    1us  763us    7ms  116ms      -
----------  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----
              capacity     operations     bandwidth    total_wait     disk_wait    syncq_wait    asyncq_wait  scrub   trim
pool        alloc   free   read  write   read  write   read  write   read  write   read  write   read  write   wait   wait
----------  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----
lwastor03   91.9T   105T    340    306  64.6M  55.3M   55ms    9ms    7ms    2ms   45us    1us    1ms    7ms   90ms      -
lwastor03a  84.7T   112T    775    305  58.8M  50.3M   34ms    9ms    3ms    1ms  427us    1us    2ms    8ms   39ms      -
lwastor03b  89.7T   107T    323    340  62.5M  54.9M   61ms    9ms    7ms    1ms   32us    1us  822us    7ms  102ms      -
----------  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----
              capacity     operations     bandwidth    total_wait     disk_wait    syncq_wait    asyncq_wait  scrub   trim
pool        alloc   free   read  write   read  write   read  write   read  write   read  write   read  write   wait   wait
----------  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----
lwastor04   88.5T   108T    208    338  15.5M  56.5M   13ms   10ms   12ms    2ms  354us    1us    2ms    8ms  623ms      -
lwastor04a  84.2T   112T    379    324  59.8M  53.9M   66ms    9ms   10ms    1ms  813us    1us    2ms    8ms  157ms      -
lwastor04b  88.1T   108T    221    341  16.9M  57.5M   13ms    9ms   12ms    2ms  421us    1us    2ms    8ms    2ms      -
----------  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----
              capacity     operations     bandwidth    total_wait     disk_wait    syncq_wait    asyncq_wait  scrub   trim
pool        alloc   free   read  write   read  write   read  write   read  write   read  write   read  write   wait   wait
----------  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----
lwastor05   88.6T   108T    239    319  19.3M  58.7M   11ms    9ms    7ms    2ms    2ms    2us    5ms    7ms    3ms      -
lwastor05a  88.6T   108T    217    318  16.6M  57.5M   12ms    9ms    8ms    2ms    2ms    2us    6ms    7ms    3ms      -
lwastor05b  90.2T   106T    208    358  16.2M  57.0M   12ms    8ms    8ms    1ms    2ms    2us    6ms    6ms    9ms      -
----------  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----
              capacity     operations     bandwidth    total_wait     disk_wait    syncq_wait    asyncq_wait  scrub   trim
pool        alloc   free   read  write   read  write   read  write   read  write   read  write   read  write   wait   wait
----------  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----
lwastor06   88.8T   108T    255    361  20.0M  59.8M   12ms    9ms   11ms    2ms  280us    1us    1ms    7ms    4ms      -
lwastor06a  89.4T   107T    235    349  18.0M  55.8M   14ms   10ms   11ms    1ms  246us    1us    1ms    9ms     1s      -
lwastor06b  88.2T   108T    341    338  58.9M  58.0M   90ms    9ms   12ms    2ms  246us    1us    2ms    8ms  244ms      -
----------  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----
              capacity     operations     bandwidth    total_wait     disk_wait    syncq_wait    asyncq_wait  scrub   trim
pool        alloc   free   read  write   read  write   read  write   read  write   read  write   read  write   wait   wait
[... cut ...]
----------  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----
lwastor08   96.1T   100T    511    334  62.7M  56.9M   72ms   28ms    9ms    2ms  384us    1us    2ms   28ms  131ms      -
lwastor08a  88.7T   108T    357    340  55.9M  60.0M   59ms   10ms   12ms    2ms  306us    1us    1ms    8ms  167ms      -
lwastor08b  88.9T   108T    349    354  52.5M  61.6M   46ms   10ms   10ms    2ms  342us    1us    1ms    8ms  118ms      -
----------  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----  -----

yupinghuang, 2024-01-18:
Since I still can't figure out access to storage node ipmi, can @kembyct take a look at the ipmi event logs and see if it's screaming at us this entire time?

Another hypothesis is power setting...I don't see signs of this but hopefully the hosts are not cutting off power to some hardware or something

caseyjlaw, 2024-01-24:
From Jack at Silicon Mechanics support:
```
Can you download and run the LSI capture script on this system so we can review the SAS logs?
https://www.broadcom.com/support/knowledgebase/1211161499563/lsiget-data-capture-script

Download the version for your operating system and controller (9500), extract, open a command prompt and run the lsiget script as Administrator/root.

This will create a support archive you can send back to me for review.

You can upload to our FTP server or send with a filesharing service. Please don't attach to an email, it will likely get blocked silently. 
```

@kembyct @rh-codebase Could you try this and send them the results? I can share the ftp site instructions separately.

rh-codebase, 2024-01-24:
@kembyct Can you do this? I'm in the middle of a bunch of test array coding now. Whichever host is having the most errors. Also, can you send a link to the summary error page you have showing how many drive errors have now been recorded as a function of host, and drive.

Is it correct that ZFS has not repaired any data associated with these errors?  If so I don't believe this is an integrity issue with the disk. Also, do read errors occur during ZFS scrubs? I personally don't think this is a disk issue since all disks have the same firmware and we are not seeing(right?) errors across all disks. Anyway, it's great SM is helping on this.

kembyct, 2024-01-24:
I can't install the package for some reasons. Is there any other way to install it.  I tired both "Broadcom Get for Linux for 96XX controllers running FW version 8.3" and "Broadcom Get for Linux for 96XX controllers running FW version 8.4". I have collected some "dmesg" meassages for each node and put it on dmesg-(date) file on each node. ZFS and dmesg buffer are not on the same page while reporting disk errors.  I wonder why. I'm researching on it. The ZFS seem fine so far.

yupinghuang, 2024-01-25:
> I can't install the package for some reasons. Is there any other way to install it. I tired both "Broadcom Get for Linux for 96XX controllers running FW version 8.3" and "Broadcom Get for Linux for 96XX controllers running FW version 8.4". 

We have a 9500 I think, not a 96XX.

> I have collected some "dmesg" meassages for each node and put it on dmesg-(date) file on each node. ZFS and dmesg buffer are not on the same page while reporting disk errors. I wonder why. I'm researching on it. The ZFS seem fine so far.

What do you mean by not being on the same page?

kembyct, 2024-01-25:
@yupinghuang The ZFS has been running most likely without errors and has an ability to repair failed disks. It's a BSD based filesystem running on a Linux system in our case. The interaction between the both seem a lil bit off so far. When the ZFS reports a disk failure, the LED light on disk slot turns red, which is accurate with the hardware performance.

kembyct, 2024-02-02:
I've downloaded the broadcom file and  run the script on lwastor06. The archive that will need to be sent back to Silicon Mechanics is in the download directory under the name:  is under the name:  lsi.linux.lwastor06.020224.224924.tgz. 
@caseyjlaw @rh-codebase @yupinghuang

caseyjlaw, 2024-02-03:
Thanks, Kemby.
I tried to log in as ubuntu@lwastor06, but am prompted for a password. Perhaps my key is not set up there.

Can you follow these instructions to upload the file to Silicon Mechanics and let me know?
```
You can upload to our FTP server or send with a filesharing service. Please don't attach to an email, it will likely get blocked silently. 

ftp.sourcecode.com
user: upload
pw: scc
```

caseyjlaw, 2024-02-03:
I know, I know. It's the end of the week, so I'll let SM handle the that.

rh-codebase, 2024-02-03:
Is this issue list public?

kembyct, 2024-02-03:
@caseyjlaw I can't connect to the ftp server for some reason. I'll send the file on lwamaas.

@rh-codebase The script needs to be run with root privilege, which I think it's not ok, so I didn't want to run on every node till SM get back to us when we send this one.