ovro-lwa/lwa-issues #504: Solar beam producing nans in one band (55-59 MHz)

dgary50, 2023-12-19:


jaycedowell, 2023-12-19:
Are they nans or zeros?

jaycedowell, 2023-12-19:
I'm looking at `/lustre/pipeline/beam02/060297_171200000000af443c8`.  If I open it and and pull out some region of interest over this frequency range:
```
import numpy as np
import h5py

# Open the file and pull out the times and frequencies contained.
h=h5py.File('060297_171200000000af443c8')
t=h['Observation1']['time'][...]
t=t['int'] + t['frac']
f=h['Observation1']['Tuning1']['freq'][...]

# Region of interest - 800 to 900 s into the file and 55 to 59 MHz
tv=np.where((t>t[0]+800) & (t<t[0]+900))[0]
fv=np.where((f>55e6) & (f<59e6))[0]

# Pull out the polarization data for the correct time range
xx=h['Observation1']['Tuning1']['XX'][tv,:]
cr=h['Observation1']['Tuning1']['XY_real'][tv,:]
ci=h['Observation1']['Tuning1']['XY_imag'][tv,:]
yy=h['Observation1']['Tuning1']['YY'][tv,:]

# Down select to the correct frequency range
xx=xx[:,fv]
cr=cr[:,fv]
ci=ci[:,fv]
yy=yy[:,fv]

# Count NaNs
m=np.zeros(xx.shape, dtype=np.int8)
m += np.where(np.isfinite(xx), 0, 1)
m += np.where(np.isfinite(yy), 0, 2)
m += np.where(np.isfinite(cr), 0, 4)
m += np.where(np.isfinite(ci), 0, 8)
print(m.shape, m.size, '&', len(np.where(m>0)[0]))

# Break down the NaNs
print(np.unique(m))
for v in np.unique(m):
    print(len(np.where(m==v)[0]))
```

I get:
```
(1558, 167) 260186 & 372
[ 0 13 14 15]
259814
118
157
97
```

Interesting things:
 * The size of this region 55 to 59 MHz region is 167 frequency channels.  That's close to the 192 channels handled by a single X-engine GPU.
   * This corresponds to roughly the F-engine channel range of 2290 to 2473.
 * XX and YY can be NaN at different times.   XX or YY being NaN leads to NaNs in Re(XY) or Im(XY) as expected.  
   * If this was an F-engine problem I might expect X and Y to be NaN at the same time.
   * The F-engine also output integers so I don't know if there really is a concept of NaN there. 
 * I can't think of how the data recorders could do this since they get packets with blocks of 96 channels from the X-engines and process the entire band at once.  I think corruption of the data would lead to a variety of strange values rather than just all NaNs and that it wouldn't be localized like this.
 * It could be an X-engine problem.  F-engine channels 2290-2473 are close to the 2288-2384 that gpu02 receives.
   * Maybe a restart of gpu02 would be a reasonable test. 
 * Within this file/time range there are two other bands that show NaNs: 25 to 27 MHz and 29 to 32 MHz.
   * These correspond to roughly F-engine channels 1040 to 1128 and 1240 to 1320.
   * These regions show a lot more structure to them than the 55 to 59 MHz area.
   * This could be RFI doing strange things.
 
Since this doesn't look to be a data recorder problem I'm de-assigning.

dgary50, 2023-12-20:
Interesting number-sleuthing, Jayce.  I am hoping that @jack-h and @caseyjlaw can look further into this.

jack-h, 2023-12-22:
Thanks for digging in to this @jaycedowell .

Nothing in the X-engine code has changed in several months, so I don't think this is a new regression. I don't recall anything of significance happening ~2 weeks ago.
I see no obvious mechanism for these NaNs -- the beamformer data casts 4-bit ints to floats, then performs beamforming matrix-multiplication. There isn't any division that should be able to result in NaNs. F-engines don't have the capacity to send NaNs, even if the data are corrupted.
If the beamforming coefficients became NaN for some reason that would explain things, but I don't see an obvious cause for this.
Is there any info I can have regarding how often / when beamforming coefficients are loaded, in case this correlates with something (I don't see how it could, given the plot in this issue).

I'm not going to be able to look at this until the other side of Christmas, but perhaps cycling the system is worthwhile in the meantime.

dgary50, 2023-12-22:
FYI, this snapshot from yesterday shows the problem is getting worse.  The SNAPs need to be power-cycled this morning (12/22 15:30 UTC), so maybe the necessary complete system reset will solve it--we'll see.
[image]

caseyjlaw, 2023-12-22:
Could we try an x-engine restart and new caltable upload first?

jaycedowell, 2023-12-22:
I think that would be helpful.  Try that and then a GPU server restart.

dgary50, 2023-12-22:
I don't know if this is relevant, but as I am poking around in the above dataset with higher resolution, I am finding some nans in other bands also.

dgary50, 2023-12-27:
This problem continues to get worse and has expanded to other bands, particularly the 25-27 MHz band.  It is now almost always occurring at certain subbands of the 45-50 MHz band, as in this plot.  The bad subbands may appear periodic in this plot, but they are not.  There are a variable number of "good" subbands between the bad ones, ranging from 6 to 9 or so.

[image]

iveydavis, 2024-01-04:
Was observing with two beams simultaneously (`con.start_dr(['dr5','dr6'], duration=17600*1e3, time_avg=256)`) and the two beams have a different number of nans (~1.1e6  in beam 6 compared to ~1.11e6 nans in beam 5)

Beam 6:
[image]

Zoom in on some of beam 6's data to show more of the structure:
[image]

The absolute difference between beam 5 and beam 6 (now on a linear instead of log scale so that 0s aren't misconstrued as nans)
[image]

And a zoom in on a bit of it to show more of the structure:
[image]

What the subtraction seems to suggest is that the exact channels that the nans concentrate at aren't the same between the two beams either. I haven't checked for any subband correlation yet.

iveydavis, 2024-01-04:
update
[image]

blue dashed lines are subband edges, black lines are channels where beam 6 had nans, and red lines are channels where beam 5 had nans. Looks like things are primarily clustered in subbands 6, 15, and 19 (and a bit in 20). I'm not sure which gpus these correspond to.

jack-h, 2024-01-07:
From 
[image]
 I believe the subband mappings are (lowest freq to highest freq):
```
server lwa-gpu01; GPU 1
server lwa-gpu01; GPU 2
server lwa-gpu02; GPU 1
server lwa-gpu02; GPU 2
server lwa-gpu03; GPU 1
server lwa-gpu03; GPU 2
server lwa-gpu04; GPU 1
server lwa-gpu04; GPU 2
...
server lwa-gpu08; GPU 1
server lwa-gpu08; GPU 2
server lwa-gpu01; GPU 1
server lwa-gpu01; GPU 2
...
server lwa-gpu08; GPU 1
server lwa-gpu08; GPU 2
```
Which I think means the problem bands don't map to a single GPU or single server (although out of the 3 servers recording NaNs in two cases both GPUs are affected) and are:
```
lwa-gpu03/2
lwa-gpu08/1
lwa-gpu08/2
lwa-gpu02/1
lwa-gpu02/2
```

dgary50, 2024-01-07:
FYI, after an xengine reset the nans may have gone away.  I am traveling but maybe Ivey can confirm. Sent from my iPhone

From
image.png (view on web)

yupinghuang, 2024-01-07:
```
(deployment) [yuping@lwacalimhead ~]$ pdsh -w lwacalim[01-08] 'ls /data??/pipeline/log | grep dr-beam-5'
<stuff>
lwacalim03: dr-beam-5.lwacalim03.log
```

note the x engine restart referenced above also involved a f engine power cycle and data recorder processes restart.

dgary50, 2024-01-07:
> (deployment) ***@***.*** ~]$ pdsh -w lwacalim[01-08] 'ls /data??/pipeline/log | grep dr-beam-5'

Actually, they seemed to go away after only an xengine restart, which I did
on Thursday after the software telecon.

iveydavis, 2024-01-07:
Just confirmed the data I collected with dr5 and dr6 both have no Nans in them (@jack-h maybe it doesn't matter at this point, but the beam pointings were separated by a few degrees)