ovro-lwa/lwa-issues #506: lwastor04a 6 degraded 2 faulted???

yupinghuang, 2023-12-26:
dmesg attached below

```
  pool: lwastor04a
 state: DEGRADED
status: One or more devices are faulted in response to persistent errors.
	Sufficient replicas exist for the pool to continue functioning in a
	degraded state.
action: Replace the faulted device, or use 'zpool clear' to mark the device
	repaired.
config:

	NAME                        STATE     READ WRITE CKSUM
	lwastor04a                  DEGRADED     0     0     0
	  raidz2-0                  DEGRADED    53     0     0
	    wwn-0x5000cca2b4cc0ee4  ONLINE       0     0     0
	    wwn-0x5000cca2b66c3790  DEGRADED    31     0     0  too many errors
	    wwn-0x5000cca2b4cbe800  FAULTED     16     0     0  too many errors
	    wwn-0x5000cca2b65a2040  ONLINE       0     0     0
	    wwn-0x5000cca2b65925e8  ONLINE       0     0     0
	    wwn-0x5000cca2b4cedcb4  DEGRADED   292     0     0  too many errors
	    wwn-0x5000cca2b663f03c  DEGRADED   576     0     1  too many errors
	    wwn-0x5000cca2b4ce3e34  FAULTED     25     0     0  too many errors
	    wwn-0x5000cca2b65d03fc  DEGRADED   664     0     0  too many errors
	    wwn-0x5000cca2b66599c0  ONLINE       0     0     0
	    wwn-0x5000cca2b6669950  DEGRADED   889     0     0  too many errors
	    wwn-0x5000cca2b4cbe9bc  DEGRADED    32     0     0  too many errors

```
[dmesg-lwastor04.txt](https://github.com/ovro-lwa/lwa-issues/files/13774123/dmesg-lwastor04.txt)

yupinghuang, 2023-12-29:
Closing in favor of https://github.com/ovro-lwa/lwa-issues/issues/503 cuz there are so many of them.

closed 2023-12-29 by yupinghuang

kembyct, 2024-01-02:
Update 
sudo zpool  status -x lwastor04a
pool 'lwastor04a' is healthy
[ubuntu@lwastor04 ~]$ sudo  zpool iostat
              capacity     operations     bandwidth 
pool        alloc   free   read  write   read  write
----------  -----  -----  -----  -----  -----  -----
lwastor04   77.9T   119T    277    314  22.8M  56.3M
lwastor04a  74.8T   122T    602    297   109M  51.1M
lwastor04b  79.3T   117T    291    326  24.1M  59.4M
----------  -----  -----  -----  -----  -----  -----

kembyct, 2024-01-08:
There is a error mistake on lwastor08.
 17          NAME                        STATE     READ WRITE CKSUM
    18          lwastor08                   DEGRADED     0     0     0
    19            raidz2-0                  DEGRADED     0     0     0
    20              wwn-0x5000cca2b6669b64  ONLINE       0     0     0
    21              wwn-0x5000cca2ab4d6368  ONLINE       0     0     0
    22              wwn-0x5000cca2b6564838  FAULTED     32     0     0  too many errors
    23              wwn-0x5000cca2b665def4  ONLINE       0     0     0
    24              wwn-0x5000cca2ab4e32f0  ONLINE       0     0     0
    25              wwn-0x5000cca2b4d4ce40  ONLINE       0     0     0
    26              wwn-0x5000cca2b4ccc05c  ONLINE       0     0     0
    27              wwn-0x5000cca2b4c9c9bc  ONLINE       0     0     0
    28              wwn-0x5000cca2b65b1100  ONLINE       0     0     0
    29              wwn-0x5000cca2b4cc0abc  ONLINE       0     0     0
    30              wwn-0x5000cca2b6509620  ONLINE       0     0     0
    31              wwn-0x5000cca2b65e247c  ONLINE       0     0     0
       
    32  errors: No known data errors

zpool clear lwastor08:
 sudo zpool status -x
  pool: lwastor08
 state: ONLINE
status: One or more devices is currently being resilvered.  The pool will
        continue to function, possibly in a degraded state.
action: Wait for the resilver to complete.
  scan: resilver in progress since Mon Jan  8 16:46:31 2024
        6.29T scanned at 0B/s, 47.4G issued at 886M/s, 94.9T total
        1.64G resilvered, 0.05% done, 1 days 07:11:52 to go
config:

        NAME                        STATE     READ WRITE CKSUM
        lwastor08                   ONLINE       0     0     0
          raidz2-0                  ONLINE       0     0     0
            wwn-0x5000cca2b6669b64  ONLINE       0     0     0
            wwn-0x5000cca2ab4d6368  ONLINE       0     0     0
            wwn-0x5000cca2b6564838  ONLINE       0     0     0  (resilvering)
            wwn-0x5000cca2b665def4  ONLINE       0     0     0
            wwn-0x5000cca2ab4e32f0  ONLINE       0     0     0
            wwn-0x5000cca2b4d4ce40  ONLINE       0     0     0
            wwn-0x5000cca2b4ccc05c  ONLINE       0     0     0
            wwn-0x5000cca2b4c9c9bc  ONLINE       0     0     0
            wwn-0x5000cca2b65b1100  ONLINE       0     0     0
            wwn-0x5000cca2b4cc0abc  ONLINE       0     0     0
            wwn-0x5000cca2b6509620  ONLINE       0     0     0
            wwn-0x5000cca2b65e247c  ONLINE       0     0     0

kembyct, 2024-01-08:
Update:
I have cleared the read error on lwastor08 pool. It's been in resilvering mode for 1 day and 4 hours. it will complete tomorow

rh-codebase, 2024-01-08:
@kembyct Is there a table somewhere you are using to keep track of how often disks, slots and hosts are having issues?  If so, are certain disks,slots and hosts more problematic than others? There may be a chance on isolating the issue if the statistics are not random.

kembyct, 2024-01-08:
No, I did not keep track of those, but I will start doing it and upload it on SharePoint.

kembyct, 2024-01-08:
Update:
The resilvering from "zpool clear lwastor08" has been complete. There is no more read error.