ovro-lwa/lwa-issues #508: New account for Andres Romero-Wolf

marinanderson, 2024-01-10:
Username: romerowo
SSH key:
 
ssh-rsa AAAAB3NzaC1yc2EAAAADAQABAAABgQC/BAR1cknpKiz+LAs8n4HPIV7Nd/CatIsRHNbztbnW7jIgaJkqeMEXSRPysY0rXtr16nIccmpjyncuL1tf3gOYQrnxUo71u/8jLuqQAiMvyBIZJVnWoP/I2eEYTLW4vIJrVDb8yQGp6igsVqYgyVGmu7jCXe7rHjj9p4yEaLJ8ovsKeymwWNTPLsUuIkMMLM4fGCV743Z/8UAR0mF7O3Cq5DFOibKT7N31wrp1LORynlY2LMywXE+ehQawk1mkN8kzjTi7q8x6cB7zwtWAiICbvhhvvK9jz7o/nDSykTcjHti2NlwdAa/pkq4K7S27oEYXSUzA7I6iuieS7+Ded8lv4mp0WU0TiWS1G0iTY2/oWp/brFyKJjC9sY9wETEGs0smd8k3U9hr1vw9mcwECMuFtrVP2j6pJA3YvG3hp5H/YDCMYgfGAN97jA/M5CQ46SJagC7xkx5PXyBghi9EuhIwI2iCBQOh6wmqbE2j+i3Z+hDIjoPmp8wJ4I/DLeqWUWk= romerowo@MT-212518

kembyct, 2024-01-15:
What's the first and last name?

caseyjlaw, 2024-01-15:
Andres Romero-Wolf

kembyct, 2024-01-15:
The account has been created.

-Kemby

marinanderson, 2024-02-01:
Hi @kembyct, Andres is still having some issues accessing `lwacalim`. He got onto `ssh.ovro.caltech.edu` without issue, but was unable to log into `lwamaas` as `ubuntu`. (His ssh-key was missing from `lwamaas`, but he still could not log on after I added it to the authorized_keys list.)

kembyct, 2024-02-01:
Hello @marinanderson 
It's done. He should be able to access it.

-Kemby