ovro-lwa/lwa-issues #509: building bifrost with cuda support on cal-im

caseyjlaw, 2024-01-17:
Issue #189 is related to this in that it includes some discussion of how to build bifrost with/without CUDA support.

@nkosogor has a working x-engine pipeline for offline beamforming. We'd like to set up an environment on cal-im to run this pipeline on demand for the imaging of triggered voltage data. This is a slight change from earlier discussions, which assumed imaging happened on the x-engines.

On cal-im, bifrost is built without CUDA support, but we'd also like to have CUDA support to use the GPUs during beamforming. @jaycedowell do you have any suggestions on how to build that without interfering with the data recorders?

yupinghuang, 2024-01-17:
Separate environment? Build your own bifrost in an environment that’s not
deployment. Or, duplicate deployment and then rebuild bifrost with cuda

caseyjlaw, 2024-01-22:
@jaycedowell Am I right in thinking that cuda support needs to be built in during c compilation? We'd like that support on one server (lwacalim02 is preferred), but it isn't clear to me that we can do that while maintaining standard data recorder as on all other nodes.

jaycedowell, 2024-01-22:
It is built during C compilation but as long as the Bifrost library is installed into a conda environment I would think that would be ok.  conda plays around with the library search paths to make sure the correct libraries are used for the relevant environment.

caseyjlaw, 2024-01-22:
Thanks. I think I have it compiling now.

closed 2024-01-22 by caseyjlaw