ovro-lwa/lwa-issues #510: access to IPMI interface on storage nodes

rh-codebase, 2024-01-18:
People need access to IPMI interface on storage nodes. Write up and distribute instructions on how to ssh and connect to the UI on the IPMI interface from a remote machine.

kembyct, 2024-01-18:
I think the issue has been the password we seem to forget.  I wonder I can reset the password in the BIOS, which will require rebooting each machine.

kembyct, 2024-01-18:
Update:
I successfully connect to ipmi host. The username is admin and the password is ADMINISTRATOR. This is the command:
"ssh -L 29443:lwastor07ipmi.mcs.pvt:443 ubuntu@lwamaas.lwa.pvt -N". Then, "https://localhost:29443" in a  browser to connect.

@rh-codebase @yupinghuang

rh-codebase, 2024-01-18:
just a reminder -- This issue list is world readable. Be careful with passwords.

kembyct, 2024-01-18:
Noted.
Thanks

rh-codebase, 2024-01-18:
The command worked for me. @yupinghuang if this works for you, please close.

caseyjlaw, 2024-01-18:
Um. I think we need to change the password now.

rh-codebase, 2024-01-18:
you don't trust people within OVRO's network? :)

caseyjlaw, 2024-01-19:
just imagine the damage I could do!

rh-codebase, 2024-01-19:
Oh yeah...  Changing it is never a bad idea. Like people say: It's no longer if you'll get hacked, it's when...